beta-tester/RPi-PXE-Server #1: Installer failing?

NorthernLoop, 2016-03-17:
Hello,

I want to say many thanks for this project it's perfect for me. I'm wondering if you can help me with the install process?

> login as: pi
> pi@192.168.0.14's password:
> 
> The programs included with the Debian GNU/Linux system are free software;
> the exact distribution terms for each program are described in the
> individual files in /usr/share/doc/*/copyright.
> 
> Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
> permitted by applicable law.
> Last login: Thu Mar 17 16:34:36 2016 from 192.168.0.107
> pi@PXE:~ $ ls
> Desktop    install-pxe-server_pass1.sh  Pictures      Templates
> Documents  install-pxe-server_pass2.sh  Public        Videos
> Downloads  Music                        python_games
> pi@PXE:~ $ bash install-pxe-server_pass1.sh
> install-pxe-server_pass1.sh: line 5: syntax error near unexpected token `newline'
> install-pxe-server_pass1.sh: line 5: `<!DOCTYPE html>'
> pi@PXE:~ $
>

NorthernLoop, 2016-03-17:
Copied the lines from script 1 and just pasted into the console window and it seems it's running. Report back after I reboot and complete part 2.

beta-tester, 2016-03-17:
at line 5 i don't see anything that could cause any error...
maybe there is something different on your system, that i expect to use to gather ip addresses.
what distro you are using?
Raspbian Jessie / Raspbian Jessie Lite?

you can try to replace the variables with fixed values
```
IP_LOCAL=$(echo $(hostname -I) | sed 's/ //g')
IP_LOCAL_=$(echo $IP_LOCAL | grep -E -o "([0-9]{1,3}[\.]){3}")
IP_LOCAL_0=$(echo $(echo $IP_LOCAL_)0)
IP_LOCAL_START=$(echo $(echo $IP_LOCAL_)200)
IP_LOCAL_END=$(echo $(echo $IP_LOCAL_)229)
IP_LOCAL_255=$(echo $(echo $IP_LOCAL_)255)
IP_ROUTER=$(grep "nameserver" /etc/resolv.conf | sed -r "s/nameserver([ ]{1,})//g")
```
with something like... (where 192.168.1.x is the local network environment)
```
IP_LOCAL=192.168.1.123
IP_LOCAL_=192.168.1.
IP_LOCAL_0=192.168.1.0
IP_LOCAL_START=192.168.1.200
IP_LOCAL_END=192.168.1.229
IP_LOCAL_255=192.168.1.266
IP_ROUTER=192.168.1.1
```

NorthernLoop, 2016-03-17:
I'm using build "2016-02-26-raspbian-jessie.img"

Will update IP addresses now. After the manual pasted script it fails to boot, I'm running headless at the moment.

NorthernLoop, 2016-03-17:
Okay the document error was because the chrome had amended the file, I can see in the downloaded file there is a load of random text. Sorry.

Downloaded the ZIP and trying again before amending the IPs.

beta-tester, 2016-03-17:
if it is not the way, how i try to figure out the network envidonment, then i don't know, what can cause that kind of error...
because there is nothing with "<DOCUMENT!>" in there...
only standard installing stuff via apt-get.

oh, wait a moment... how did you copiesd the scripts from git hub...? maybe the files are in html format instead of plain text...

beta-tester, 2016-03-17:
try to download the zip file...
https://github.com/beta-tester/install-pxe-server/archive/master.zip

NorthernLoop, 2016-03-17:
Applied the Zip file an installer runs. On reboot it doesn't up back to my IP (192.168.0.14)

> 
IP_LOCAL=192.168.0.14
**IP_LOCAL_=192.168.0.**
IP_LOCAL_0=192.168.0.0
IP_LOCAL_START=192.168.0.200
IP_LOCAL_END=192.168.0.229
IP_LOCAL_255=192.168.0.266
IP_ROUTER=192.168.0.1
> 

Can I query the address in bold?

beta-tester, 2016-03-17:
the adresses are ok, eycept of the ```IP_LOCAL_255=192.168.0.266``` that sould be ```IP_LOCAL_255=192.168.0.255``` but it was my mistake... :oops:

beta-tester, 2016-03-17:
hmm... if you are doing everything via SSH connection,
i think you will run into problems after rebooting after the first script.
there is a bug in systemd/dhcpd dependency ordering. that can cause the system to loose the network connection until you add the workaround for that, that i put unfortunately to the second script.
maybe i should put the workaround to the first script or to both... :thought_balloon:

NorthernLoop, 2016-03-17:
Yeah I'm running without screen. Though I have just connected a 3.5" screen to see what is happening.

Running script without amending IPs.
PXE IP is 192.168.0.14
Router 192.168.0.1
Sub 255.255.255.0

NorthernLoop, 2016-03-17:
Okay after running unedited script, part 1 installs and then I reboot. On reboot stopped on 

> Mounting Configuration File System

>

beta-tester, 2016-03-17:
damn...
if you dont have an USB stick thie the correct name, then your system stucks here...
please delete or adjust the entry in the /etc/fstab file to
```
LABEL=PXE-Server  /media/server/  auto  defaults,noatime,auto,x-systemd.automount  0  0

```

NorthernLoop, 2016-03-17:
I don't have a stick in the device but I do have a portable 2.5" 1TB drive I was planning to use. I have loads of images I want to PXE.

beta-tester, 2016-03-17:
USB-Stick or HDD doesn't matter. that fstab entry expect a filesystem with a label "PXE-Server", you can adjust to your needs.

NorthernLoop, 2016-03-17:
OK the Pi fails to power my portable HDD, found another and boots. Running second script now.

I'm using 128Gb microSD card. I'm guessing the ISO images are meant to be stored on the USB drive and not the microsd?

beta-tester, 2016-03-17:
thank you for testing and telling me the problems you got... i changed the first script to do not run into that problems now.

the second script as is, is currently made to store all stuff to SD card.
 to store the iso's to the external drive, you have to adjust the script.
also to download/use other iso's as in the script, you have to adjust the script.

NorthernLoop, 2016-03-17:
Great, ever time I try to amend the sh script and upload using WinSCP it reports an error. Guessing its adding some formatting into the file?

Downloading new version and I'll report back. 

To begin with all the ISO images on the SD card for me as well.

beta-tester, 2016-03-17:
the second script i didn't touched... i only changed the first script!

changing script under windows and copying to linux will caus errors because of linux uses LF and windows CR + LF
LF = line feed
CR = carriage return

notepad++ from https://notepad-plus-plus.org/ is a good text editor that can handle/convert files from/to linux/windows.

NorthernLoop, 2016-03-18:
Many thanks for your support on this.

I have done a clean install after it still freezes on boot when no USB is attached.

beta-tester, 2016-03-18:
if it still freeze, then remove/comment out the whole mount entry for the external drive from */etc/fstab* file, then is shoudn't freeze.
``` ### LABEL=PXE-Server  /media/server/  auto  defaults,noatime,auto,x-systemd.automount  0  0```

after that, if it still freeze then it is not the mount entry...

closed 2016-04-21 by beta-tester