beta-tester/RPi-PXE-Server #2: Error on pass2.sh

Comm4nd0, 2017-11-18:
Error: ./install-pxe-server_pass2.sh: 784: [: http://releases.ubuntu.com/17.10/ubuntu-17.10-desktop-amd64.iso: unexpected operator

Can't give any more info than that. 

Any ideas?

beta-tester, 2017-11-23:
sorry, i saw your issue lately...

what shell you are using (bash)?

how does the following script lines looks like in your file?
line 111:
`UBUNTU_LTS_X64=ubuntu-lts-x64`

line 112:
`UBUNTU_LTS_X64_URL=http://releases.ubuntu.com/16.04.3/ubuntu-16.04.3-desktop-amd64.iso`

line 773:
` local URL=$2`

line 784:
`if [ "$URL" == "" ]; then`

line 1283:
`handle_iso $UBUNTU_X64 $UBUNTU_X64_URL;`

i just cloned the script and executes the script without any issues.
RPi2, Raspbian Stretch.
`$ bash install-pxe-server_pass2.sh`

Comm4nd0, 2017-11-25:
interesting, i ran it like this:
bash install-pxe-server_pass2.sh

and it seems to be working this time. before i ran it like this:

./install-pxe-server_pass2.sh

maybe that is the problem?

Comm4nd0, 2017-11-25:
So it's been sitting on this for an hour:
sudo bash install-pxe-server_pass2.sh 
setup variables

eth0 is used as primary networkadapter for PXE
192.168.0.53 is used as primary IP address for PXE
12345678 is used as SN for RPi3 network booting

handle_dnsmasq()
handle_samba()
handle_optional()
handle_dhcpcd()
    a non-stretch os detected
handle_iso(ubuntu-x64)

beta-tester, 2017-11-26:
i have not that shell knowledge, so i don't know the differences between **_sh_** and **_bash_**.
i guess there are different in expanding variables inside of quotes.

PS.: no "**_sudo_**" needed (or allowed)
run the script without **_sudo_**, because in the script there are the sudo commands in.
it is not tested to run the script from within sudo context - no idea, if it has side effects or makes trouble.

Comm4nd0, 2017-11-29:
So i'm running it with this command:
`bash install-pxe-server_pass2.sh`

and it's still sitting on:
`setup variables

eth0 is used as primary networkadapter for PXE
192.168.0.53 is used as primary IP address for PXE
12345678 is used as SN for RPi3 network booting

handle_dnsmasq()
handle_samba()
handle_optional()
handle_dhcpcd()
    a non-stretch os detected
handle_iso(ubuntu-x64)
`

Very strange, i can't think what the differences might be between our systems. I'm using a raspi3 which is up to date.

Do you know what line is being run at the point of where i am? I might be able to trouble shoot by running the commands manually.

beta-tester, 2017-11-30:
yes, at this point it is at line 777 or beyond.
normally it will:
1. create directories if not already exist,
2. (at line 781) unexport nfs,
3, (at line 782) unmount iso,
4. ... this point you have not reached, because you should see another message or status on the output.

step 2 or 3 are the candidats,
but even the nfs exports and mounted iso may not existing at this moment, because of you are running that script the very first time and the iso file is not downloaded yet, it should not block...

PS.: i updated the script from `#!/bin/sh` to `#!/bin/bash`, maybe that will fix the issue you also had with executing without **_bash_** in front of the script.

Comm4nd0, 2017-11-30:
i believe it's jessie. is this ok?

I've already taken out most of the stderr redirects. the only thing i get is an error on the umount because it's not already mounted. but i'm sure that is ok.

beta-tester, 2017-11-30:
that's very strange.

i think jessie should not make trouble. maybe **_rsync_** is not installed there, but then you should see an error, when this is the stopper.

when you get the error on unmounting (that is normal), then you passed line 781 and line 782.
next output would an echo after some if statements.
when you comment out line 795 to line 805 (this will try to find and copy the iso from an attached USB stick. maybe that **_grep_** or **_rsync_** is failing)

i am still wondering at which line it is stucking.

Comm4nd0, 2017-11-30:
i commented out lines 795 to line 805 and it's now rsyncing correctly. No idea why at the moment.

Comm4nd0, 2017-11-30:
ok so now it's getting stuck at:

```
./install-pxe-server_pass2.sh 
setup variables

eth0 is used as primary networkadapter for PXE
192.168.0.53 is used as primary IP address for PXE
12345678 is used as SN for RPi3 network booting

handle_dnsmasq()
handle_samba()
handle_optional()
handle_dhcpcd()
    a non-stretch os detected
handle_iso(ubuntu-x64)
    download iso image
--2017-11-30 21:24:17--  http://releases.ubuntu.com/17.10/ubuntu-17.10-desktop-amd64.iso
Resolving releases.ubuntu.com (releases.ubuntu.com)... 91.189.88.160, 2001:67c:1560:8001::7
Connecting to releases.ubuntu.com (releases.ubuntu.com)|91.189.88.160|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 1501102080 (1.4G) [application/x-iso9660-image]
Saving to: ‘/srv/iso/ubuntu-x64.iso’

/srv/iso/ubuntu-x64.iso       100%[===================================================>]   1.40G   915KB/s   in 39m 20ss

2017-11-30 22:03:37 (621 KB/s) - ‘/srv/iso/ubuntu-x64.iso’ saved [1501102080/1501102080]

    create nfs folder
    add iso image to fstab
    add nfs folder to exports
handle_pxe()
    copy win-pe stuff
```
I can now see the PXE boot from a VM. but if i select ubuntu after a while i get `NFS over TCP not available from 192.168.0.53`

Comm4nd0, 2017-11-30:
After doing some digging it seems to be getting stuck at line 1158. I've tried viewing the contents of /media/server but there is nothing there. maybe it's only there while the script is running?

beta-tester, 2017-12-01:
it  is normal, that `/media/server` is empty, when you don't have an usb stick attached with `LABEL=PXE-Server` at boot time. but that shouldn't make trouble.

today, i will install raspbian jessie and run my script without an attached usb-stick, to see, if i can reproduce that behavior you have...

please stay tuned...

beta-tester, 2017-12-01:
oops... i can fully reproduce your behavior!
your hint with the empty `/media/server` mountpoint was very good.
each file access to that mountpoint tries to mount the not existing source (usb-stick). and exactly this will block the system.
i think the behavior under raspbian stretch is different there. (i have to test this as well).

i will update the script later the day (i am currently not sure how to do best).

but what you can do immediately is to unmount the mountpoint where no usb-stick is behind with:
```
sudo umount -f /media/server
```
and remove the whole entry in `/etc/fstab`
```
LABEL=PXE-Server ...
```
to get rid of the mountpoint.

thank you very much for your report.

Comm4nd0, 2017-12-01:
Not a problem! i'm glad we've manages to get to the bottom of it.

On a side note, i'm actually a Python, Bash and Ansible developer as my day job. The reason i'm trying to utilise what you have already made is because i want to make a device(raspPi) that will connected to a computer and image the computer using your PXE boot install and then automatically run an Ansible playbook to get the machine to the desired state.

beta-tester, 2017-12-01:
do you have knowledge about Red Hat/Fedoar/CentOS ?
i still have trouble to get it working to PXE boot directly from its LiveDVD...
but i don't gent response to all my questions about that on Fedore and CentOS forum.

maybe you can fogure out what pxe menu entry APPEND parameter are needed to pxe boot those systems.

closed 2017-12-01 by beta-tester

Comm4nd0, 2017-12-01:
So i use Fedora as my main laptop. i'd be happy to try and work it out for you.