beta-tester/RPi-PXE-Server #3: Can't get Windows PE boot to work..

TheHerodes, 2017-11-24:
I filled in the script where to find the PE.iso
The script made a label in de menu.
But when trying to boot to PE it shows an error no such file or directory
When I change the location /iso/ to something else and move the PE.iso into this new location  I get the same error.
I'm not used to linux so please could you help me out?

beta-tester, 2017-11-26:
did you have a WinPE iso file?
and is the file called "win-pe-x86.iso" ?

TheHerodes, 2017-11-26:
Yes

beta-tester, 2017-11-27:
and what about the script changes you made?

TheHerodes, 2017-11-28:
I deleted the # for the script line and changed url to the location of the
iso.
That didn't work out because the script didn't copy the iso to the usb
stick.
So i did that by hand with sudo cp.

beta-tester, 2017-11-28:
ok,
- your  boot entry appears as "**_Windows PE x86 (ISO)_**" in the PXE menu and this entry makes trouble.
when you choose that boot option, you will see (maybe for a very short time):
```
Loading: /memdisk ... ok
Loading: /iso/win-pe-x86.iso ... failed: no such file or directory

```
- your pxe menu entry looks like
```
LABEL  Windows PE x86 (ISO)
    KERNEL /memdisk
    APPEND iso
    INITRD /iso/win-pe-x86.iso
```
- your **_win-pe-x86.iso_** file is located at **_/srv/iso/win-pe-x86.iso_** with at least read access to everybody.
```
$ ls -la /srv/iso/
-rw-rw-r-- 1 root root  346060800 Nov 27 21:51 win-pe-x86.iso
-rw-rw-r-- 1 root root          1 Nov 27 21:52 win-pe-x86.url
...
```
- you have a symbolic link at **_/srv/tftp/menu-bios/iso_** that is pointing to folder **_/srv/iso_** with at least read and execute access to everybody
```
$ ls -la /srv/tftp/menu-bios/
lrwxrwxrwx 1 root root     9 Nov 27 21:50 iso -> /srv/iso/
lrwxrwxrwx 1 root root    25 Nov 27 21:50 memdisk -> /usr/lib/syslinux/memdisk
...
```
- you still have trouble to boot into **_win-pe-x86.iso_**, even when you copied that file by hand to its correct location and applied the file access permissions to it.

so far, are my all assumptions correct?

is something different, then please give me, what you exactly have and where exactly all your file paths are.

TheHerodes, 2017-11-29:
I think it works now.
Thanks for your support.
I changed some things so i don't know exactly what the problem was.(change
group and user on the iso file & I made the symbolic link manually)
I'm still a noob in Linux (Windows user)

Greetings,

Peter

beta-tester, 2017-11-29:
well, then i close that issue.

closed 2017-11-29 by beta-tester

gh0st, 2019-04-25:
I'm experiencing the same issue. I don't think you posted what resolved the problem in the thread above.

beta-tester, 2019-04-25:
hello @gh0st ,
what Windows PE version do you use and where did you get it or how did you created it?

gh0st, 2019-04-25:
You know what? Let me take this back to the drawing board. I literally took a stock Windows 7 `.iso` and tried to pxe serve it without PE prepping it first.

beta-tester, 2019-04-25:
you mean you took a windows (7) installation iso?
this is too big to boot as memdisk.
you need a windows PE iso (from 7 or 10) that is not bigger than 500 MB otherwise the memdisk method is not working.

beta-tester, 2019-04-25:
did you also checked https://github.com/beta-tester/RPi-PXE-Server/issues/3#issuecomment-347498950 ?

gh0st, 2019-04-26:
I looked at #3 and tried it before mentioning anything. Yes I took a win7 install ISO. I'll try the win PE iso.

beta-tester, 2019-04-26:
the problem with memdisk is, that the bootloader will load the iso image into the RAM and starts the boot file like a DVD. but the booted system does not recognized the ISO image in the RAM.
no drive letter, no disk lable, no physical device is assigned for windows.
so once the installation system is booted, it can not access to the required files.

i think that is the reason for the "file not found error" there.

see also https://wiki.syslinux.org/wiki/index.php?title=MEMDISK

then you have to boot a win PE environment, start the network interface `wpeutil initializenetwork`, mount the SMB (samba) share `net use N: \\PXE-SERVER\srv guest /USER:guest` for the installation files and run the `N:\windown7-media\setup.exe` of windows to install windows.

the PXE bootoption would be to do not use memdisk, but that would take much more handmade action to prepare windows BCD boot menu entries, the reqired boot files and the installation files.

beta-tester, 2019-04-26:
just saw something...
try to use option `raw`. maybe that will do the trick.
```
LABEL  Windows PE x86 (ISO)
    KERNEL /memdisk
    APPEND iso raw
    INITRD /iso/win-pe-x86.iso
```

gh0st, 2019-04-26:
Ok I'll try this too. I saw that in someone else's pxe server implementation.

gh0st, 2019-05-17:
Ok I changed my `...include-menu.sh` line 775 to read
```
APPEND iso raw
```
I run `bash install-pxe-server_pass2.sh` after the change. Try a pxe boot again, see _Windows PE x86 (ISO)_ in my pxe boot menu. When I select it and hit enter though, it hangs with
```
Loading memdisk... ok
Loading http://192.168.1.109/iso/winpe-x86.iso...
```

I think I will try creating that winpe image from the win7.iso I have.

beta-tester, 2019-05-17:
@gh0st , i recomment to create a Windows 10 PE iso. it has more modern device drivers and features.
i never tried a windows 7 one.
you can download the **Windows Assessment and Deployment Kit (ADK)** and create a Windows 10 PE iso by following this link:
[Create bootable WinPE media](https://docs.microsoft.com/de-de/windows-hardware/manufacture/desktop/winpe-create-usb-bootable-drive)

beta-tester, 2019-05-17:
when i create a win pe iso, i always install the following add-ons on top of the default iso:
```
WinPE-MDAC.cab
WinPE-FMAPI.cab
WinPE-WiFi-Paket.cab
WinPE-WMI.cab
WinPE-Scripting.cab
WinPE-NetFX.cab
WinPE-PowerShell.cab
WinPE-DismCmdlets.cab
WinPE-SecureBootCmdlets.cab
WinPE-StorageWMI.cab
WinPE-HTA.cab
WinPE-SecureStartup.cab
WinPE-EnhancedStorage.cab
```

and in case you use a german keyboard:
```
/Set-UILang:de-de
/Set-SysLocale:de-de
/Set-UserLocale:de-de
/Set-InputLocale:de-de
/Set-AllIntl:de-de
/Set-SKUIntlDefaults:de-de
/Set-TimeZone:"W. Europe Standard Time"
```

see also:
[WinPE Optional Components (OC) Reference](https://docs.microsoft.com/en-us/windows-hardware/manufacture/desktop/winpe-add-packages--optional-components-reference)

beta-tester, 2020-12-22:
added a tiny wiki how to create a basic WinPE iso file
https://github.com/beta-tester/RPi-PXE-Server/wiki/Create-a-WinPE-ISO