beta-tester/RPi-PXE-Server #5: Store img files on the USB?

HarvsG, 2018-05-22:
I dont have an SD card big enough to store all the img files, can I store them on the USB instead, if so how?

beta-tester, 2018-05-23:
to store all IMG/ISO/... files to an external USB memory Stick or HDD, the best is to format the stick with a ext4 filesystem.
give the filesystem the label "PXE-Server" then is should get mounted by the install script automatically. then create a folder to the usb stick and change the install script to use that folder for all the stuff...

e.g.
(expecting that you have a virgine Raspbian-Stretch image on your SD card and an empty USB memory stick/HDD on your RPi):
1. `sudo mkfs.ext4 -L PXE-Server /dev/sdX` where **_sdX_** the USB drive/partition is.
2. `bash install-pxe-server_pass1.sh` will create an **/etc/fstab** entry for automount the filesystem with label "PXE-Server" 
3. `sudo mkdir /media/server/srv` to create a folder as destination, where all the download stuff will be stored and populated for PXE boot later on.
4. in the script "install-pxe-server_pass2.sh"
change `DST_ROOT=/srv` to `DST_ROOT=/media/server/srv`
5. `bash install-pxe-server_pass2.sh`...

this should do the job.

if the label of the USB file system is a different one than "PXE-Server", then you have to add an additional **/etc/fstab** entry or adjust the existing to mount the file system automatically at each boot. and you also may have to adjust the `DST_ROOT=/srv` to where ever your mountpoint is and what ever name your folder there has.

HarvsG, 2018-05-25:
Thank you I did this and had some success! However the 28gb USB filled quite quickly. On further inspection I noticed some img/iso/... Were in both `media/sever/backup` and `media/sever/srv` is this why it's filled quickly, and is there a good reason for the duplication?

I am having issue with UEFI over IPv4 booting. The DHCP etc seems to be working fine but then I get a filsysi error

On an unrelated topic I thought I'd mention a few things
1. I did this install on a raspberry pi that is also running OpenVPN, PiHole and DNS over HTTPS (I had to manage a few DNSmasq conflicts) 
2. I am not running a DNSmasq proxy DHCP server, which allows my home router to run the DHCP service with no pxe/tfttp/option 66 configuration (it can't).

beta-tester, 2018-05-26:
sorry, i forgot to tell you.

yes, then the data are now stored twice.
once to the **srv** folder and then as backup to the **backup** folder.

i have that backup option, because of my poor internet speed. so, i only have to download changed images when i start from scratch, and the images i already downloaded before i take from the backup. 

you can try to delete the whole **backup** folder.
as soon there is no backup folder available, the script should not backup the images anymore.

**UEFI** isn't tested, because i still have BIOS only hardware available to test.

HarvsG, 2018-05-27:
Moved the (U)EFI discussion to #6

beta-tester, 2018-05-28:
is it working from USB stick now, can i close this issue?

HarvsG, 2018-05-28:
Yes. I will close the issue. I just deleted the backup file

closed 2018-05-28 by HarvsG