beta-tester/RPi-PXE-Server #6: Boot not working for EFI

HarvsG, 2018-05-27:
As referenced in #5, comment 4. I cannot get IPv4 UEFI boot to work. Initially, I suffered an error prompt from my BIOS "`Selected boot image did not authenticate`" then after disabling secure boot, I got an error "`The selected boot device failed to load`"
"`The selected boot device failed to load`". Suggests the issue here is two-fold, firstly with secure boot an then with the PXE configuration. 

After you (@beta-tester) pointed this out "did you removed the '#' in front of the efi32 and efi64 stuff i comment out?", I did so and that is what got me the "`The selected boot device failed to load`" error, initially I was only getting the "`The selected boot device failed to load`" error, even with secure boot enabled.

I am going to try with an iPXE device so I can get more detailed error reports and will update.

I am yet to test this "what you also can try is to offer the UEFI computer the UEFI compatible boot-manager-file directly (in dnsmasq) instead of offering the pxe-menu file. but then you don't have the choice anymore to select different OS'es for UEFI computers."

HarvsG, 2018-05-27:
So I go the output of the logs from dnsmasq to see what the server was doing:
`May 27 22:44:31 dnsmasq-tftp[3761]: sent /media/server/srv/tftp/menu-efi64/syslinux.0 to 192.168.0.31 `
With repeated attempts, that was what I got nearly every time, so the DHCP must be working fine and it was sending syslinux.0 happily.

Once or twice i got the below errors, although not that often.
```
May 27 22:44:31 dnsmasq-tftp[3761]: failed sending /media/server/srv/tftp/menu-efi64/syslinux.0 to 192.168.0.31
May 27 22:44:31 dnsmasq-tftp[3761]: error 8 User aborted the transfer received from 192.168.0.31       
```

Going from these rescources [[1](https://wiki.ubuntu.com/UEFI/PXE-netboot-install)],[[2](https://wiki.ubuntu.com/UEFI/SecureBoot/PXE-IPv6?action=show&redirect=UEFI%2FSecureBoot-PXE-IPv6)] it seems a .efi file is expected, and if it is to be secure boot compatible, a signed one at that. It seems grub would do, but does it suppport multiboot?

HarvsG, 2018-05-28:
So I have made a few changes: on [this fork](https://github.com/HarvsG/RPi-PXE-Server/tree/95be45373432c5714e57e0e3e1a1ba85c5fd44d9). 
They are:
1 .  Adds install for [<code>syslinux-efi</code>](https://packages.debian.org/sid/syslinux-efi) package in the first script. https://github.com/HarvsG/RPi-PXE-Server/commit/90bd6277f57bdf914d743fc5b820ce8d35176fca
2.  Add a symlink for the [<code>.efi</code>](https://packages.debian.org/sid/all/syslinux-efi/filelist) file that  [<code>syslinux-efi</code>](https://packages.debian.org/sid/syslinux-efi) installs. https://github.com/HarvsG/RPi-PXE-Server/commit/be38a7341801de4d1e410763b88203b7c5faebc0
3. Make to relevant changes in the dnsmasq config https://github.com/HarvsG/RPi-PXE-Server/commit/95be45373432c5714e57e0e3e1a1ba85c5fd44d9 & https://github.com/HarvsG/RPi-PXE-Server/commit/c695cdf95ab559be6639e7902de0d75562ab86eb

It gets me past the error messages, to a frozen screen that tells me the IP of my pc and says that it is looking for cached packets.
```
May 27 23:50:07 dnsmasq-dhcp[5725]: DHCP, proxy on subnet 192.168.0.35
May 27 23:50:07 dnsmasq-tftp[5725]: TFTP enabled
May 27 23:50:13 dnsmasq-dhcp[5725]: 3154604121 available DHCP subnet: 192.168.0.35/255.255.255.0
May 27 23:50:13 dnsmasq-dhcp[5725]: 3154604121 available DHCP range: 192.168.0.150 -- 192.168.0.200
May 27 23:50:13 dnsmasq-dhcp[5725]: 3154604121 vendor class: PXEClient:Arch:00007:UNDI:003016
May 27 23:50:13 dnsmasq-dhcp[5725]: 3154604121 PXE(eth0) 6c:c2:17:5f:a1:30 proxy
May 27 23:50:13 dnsmasq-dhcp[5725]: 3154604121 tags: x64_UEFI, eth0
May 27 23:50:13 dnsmasq-dhcp[5725]: 3154604121 next server: 192.168.0.35
May 27 23:50:13 dnsmasq-dhcp[5725]: 3154604121 broadcast response
May 27 23:50:13 dnsmasq-dhcp[5725]: 3154604121 sent size:  1 option: 53 message-type  2
May 27 23:50:13 dnsmasq-dhcp[5725]: 3154604121 sent size:  4 option: 54 server-identifier  192.168.0.35
May 27 23:50:13 dnsmasq-dhcp[5725]: 3154604121 sent size:  9 option: 60 vendor-class  50:58:45:43:6c:69:65:6e:74
May 27 23:50:13 dnsmasq-dhcp[5725]: 3154604121 sent size: 17 option: 97 client-machine-id  00:35:43:44:33:34:35:37:5a:4d:4d:6c:c2:17...
May 27 23:50:17 dnsmasq-dhcp[5725]: 3154604121 available DHCP subnet: 192.168.0.35/255.255.255.0
May 27 23:50:17 dnsmasq-dhcp[5725]: 3154604121 available DHCP range: 192.168.0.150 -- 192.168.0.200
May 27 23:50:17 dnsmasq-dhcp[5725]: 3154604121 vendor class: PXEClient:Arch:00007:UNDI:003016
May 27 23:50:17 dnsmasq-dhcp[5725]: 4203663260 available DHCP subnet: 192.168.0.35/255.255.255.0
May 27 23:50:17 dnsmasq-dhcp[5725]: 4203663260 available DHCP range: 192.168.0.150 -- 192.168.0.200
May 27 23:50:17 dnsmasq-dhcp[5725]: 4203663260 vendor class: PXEClient:Arch:00007:UNDI:003016
May 27 23:50:17 dnsmasq-dhcp[5725]: 4203663260 PXE(eth0) 6c:c2:17:5f:a1:30 proxy
May 27 23:50:17 dnsmasq-dhcp[5725]: 4203663260 tags: x64_UEFI, eth0
May 27 23:50:17 dnsmasq-dhcp[5725]: 4203663260 bootfile name: menu-efi64/syslinux.efi
May 27 23:50:17 dnsmasq-dhcp[5725]: 4203663260 server name: 192.168.0.35
May 27 23:50:17 dnsmasq-dhcp[5725]: 4203663260 next server: 192.168.0.35
May 27 23:50:17 dnsmasq-dhcp[5725]: 4203663260 sent size:  1 option: 53 message-type  5
May 27 23:50:17 dnsmasq-dhcp[5725]: 4203663260 sent size:  4 option: 54 server-identifier  192.168.0.35
May 27 23:50:17 dnsmasq-dhcp[5725]: 4203663260 sent size:  9 option: 60 vendor-class  50:58:45:43:6c:69:65:6e:74
May 27 23:50:17 dnsmasq-dhcp[5725]: 4203663260 sent size: 17 option: 97 client-machine-id  00:35:43:44:33:34:35:37:5a:4d:4d:6c:c2:17...
May 27 23:50:18 dnsmasq-tftp[5725]: error 8 User aborted the transfer received from 192.168.0.31
May 27 23:50:18 dnsmasq-tftp[5725]: failed sending /media/server/srv/tftp/menu-efi64/syslinux.efi to 192.168.0.31
May 27 23:50:18 dnsmasq-tftp[5725]: sent /media/server/srv/tftp/menu-efi64/syslinux.efi to 192.168.0.31
```

Baisically what I think is going on here is that these lines are copy-paste from the BIOS section, but should be a whole new set of files, unrelated to the BIOS ones installed by [<code>syslinux-common</code>](https://packages.debian.org/sid/syslinux-common)https://github.com/beta-tester/RPi-PXE-Server/blob/383b85432b3b18ac43de0eb36f210eb4d1d1e229/install-pxe-server_pass2.sh#L1545-L1553

HarvsG, 2018-05-28:
so I downloaded the [syslinux v6.04-pre1](https://mirrors.edge.kernel.org/pub/linux/utils/boot/syslinux/Testing/6.04/) and manually updated just the .efi file and got a new error

```
>>>Start PXE over IPv4, Press [ESC] to EXIT...
    Station IP address is 192.168.0.31

    Server IP address is 192.168.0.35
    NBP filename is menu-efi64/syslinux.efi
    NBP filesize is 176208 Bytes
  Downloading NBP file...

    Succeed to download NBP file.
Getting cached packet
My IP is 192.168.0.31
Unable to locate configuration file
_
```

beta-tester, 2018-05-29:
your changed are integrated now.
see https://github.com/beta-tester/RPi-PXE-Server/commit/44d581a83f65099e87ec316f4c49dae3271e52ac

again, thank you very much for your help.

closed 2018-05-29 by beta-tester

squadramunter, 2019-04-23:
I have tried to boot with UEFI mode in BIOS settings but it did not work. I use Reaktek controller to boot from it. My laptop is Asus K501LX. Booting from Legacy (CSM) does work but UEFI doesn't.