beta-tester/RPi-PXE-Server #8: Raspberry Pi3 not booting

HarvsG, 2018-05-28:
I have a raspberry pi 3 configured for network booting and I re-ran the installation script with the serial number in. When I try and boot I get the below server logs.
```
May 28 12:23:33 dnsmasq-dhcp[676]: 653460281 available DHCP subnet: 192.168.0.35/255.255.255.0
May 28 12:23:33 dnsmasq-dhcp[676]: 653460281 available DHCP range: 192.168.0.150 -- 192.168.0.200
May 28 12:23:33 dnsmasq-dhcp[676]: 653460281 vendor class: PXEClient:Arch:00000:UNDI:002001
May 28 12:23:33 dnsmasq-dhcp[676]: 653460281 PXE(eth0) b8:27:eb:41:4a:9d proxy
May 28 12:23:33 dnsmasq-dhcp[676]: 653460281 tags: ARM_RPI3, UUID_RPI3, ARCH_0, client_is_a_pi, eth0
May 28 12:23:33 dnsmasq-dhcp[676]: 653460281 reply delay: 2
May 28 12:23:35 dnsmasq-dhcp[676]: 653460281 bootfile name: bootcode.bin
May 28 12:23:35 dnsmasq-dhcp[676]: 653460281 broadcast response
May 28 12:23:35 dnsmasq-dhcp[676]: 653460281 sent size:  1 option: 53 message-type  2
May 28 12:23:35 dnsmasq-dhcp[676]: 653460281 sent size:  4 option: 54 server-identifier  192.168.0.35
May 28 12:23:35 dnsmasq-dhcp[676]: 653460281 sent size:  9 option: 60 vendor-class  50:58:45:43:6c:69:65:6e:74
May 28 12:23:35 dnsmasq-dhcp[676]: 653460281 sent size: 17 option: 97 client-machine-id  00:44:44:44:44:44:44:44:44:44:44:44:44:44...
May 28 12:23:35 dnsmasq-dhcp[676]: 653460281 sent size: 47 option: 43 vendor-encap  06:01:03:0a:04:00:50:58:45:08:07:80:00:01...
May 28 12:23:35 dnsmasq-tftp[676]: file /media/server/srv/tftp/bootsig.bin not found
May 28 12:23:35 dnsmasq-tftp[676]: sent /media/server/srv/tftp/bootcode.bin to 192.168.0.25
May 28 12:23:35 dnsmasq-tftp[676]: file /media/server/srv/tftp/<MyRPI_SN>/start.elf not found
May 28 12:23:35 dnsmasq-tftp[676]: file /media/server/srv/tftp/autoboot.txt not found
May 28 12:23:35 dnsmasq-tftp[676]: file /media/server/srv/tftp/config.txt not found
May 28 12:23:35 dnsmasq-tftp[676]: file /media/server/srv/tftp/recovery.elf not found
May 28 12:23:35 dnsmasq-tftp[676]: file /media/server/srv/tftp/start.elf not found
May 28 12:23:35 dnsmasq-tftp[676]: file /media/server/srv/tftp/fixup.dat not found
```
A few things I've also noticed:
* The green light flashes on the Pi, so perhaps it has booted, but I just cant SSH into it... (it doesn't respond to pings though)
* `tftp/boot `is empy
* `nfs/rpi-<MyRPI_SN>-root` is empty

HarvsG, 2018-05-28:
doing `sudo wget https://github.com/raspberrypi/firmware/raw/stable/boot/start.elf` whilst in `<MyRPI_SN>/` got me as far as:
```
May 28 12:48:15 dnsmasq-dhcp[676]: 653460281 available DHCP subnet: 192.168.0.35/255.255.255.0
May 28 12:48:15 dnsmasq-dhcp[676]: 653460281 available DHCP range: 192.168.0.150 -- 192.168.0.200
May 28 12:48:15 dnsmasq-dhcp[676]: 653460281 vendor class: PXEClient:Arch:00000:UNDI:002001
May 28 12:48:15 dnsmasq-dhcp[676]: 653460281 PXE(eth0) b8:xx:xx:xx:xx:xx proxy
May 28 12:48:15 dnsmasq-dhcp[676]: 653460281 tags: ARM_RPI3, UUID_RPI3, ARCH_0, client_is_a_pi, eth0
May 28 12:48:15 dnsmasq-dhcp[676]: 653460281 reply delay: 2
May 28 12:48:17 dnsmasq-dhcp[676]: 653460281 bootfile name: bootcode.bin
May 28 12:48:17 dnsmasq-dhcp[676]: 653460281 broadcast response
May 28 12:48:17 dnsmasq-dhcp[676]: 653460281 sent size:  1 option: 53 message-type  2
May 28 12:48:17 dnsmasq-dhcp[676]: 653460281 sent size:  4 option: 54 server-identifier  192.168.0.35
May 28 12:48:17 dnsmasq-dhcp[676]: 653460281 sent size:  9 option: 60 vendor-class  50:58:45:43:6c:69:65:6e:74
May 28 12:48:17 dnsmasq-dhcp[676]: 653460281 sent size: 17 option: 97 client-machine-id  00:44:44:44:44:44:44:44:44:44:44:44:44:44...
May 28 12:48:17 dnsmasq-dhcp[676]: 653460281 sent size: 47 option: 43 vendor-encap  06:01:03:0a:04:00:50:58:45:08:07:80:00:01...
May 28 12:48:17 dnsmasq-tftp[676]: file /media/server/srv/tftp/bootsig.bin not found
May 28 12:48:17 dnsmasq-tftp[676]: sent /media/server/srv/tftp/bootcode.bin to 192.168.0.25
May 28 12:48:17 dnsmasq-tftp[676]: file /media/server/srv/tftp/37414a9d/autoboot.txt not found
May 28 12:48:17 dnsmasq-tftp[676]: error 0 Early terminate received from 192.168.0.25
May 28 12:48:17 dnsmasq-tftp[676]: failed sending /media/server/srv/tftp/<MyRPI_SN>/start.elf to 192.168.0.25
May 28 12:48:17 dnsmasq-tftp[676]: sent /media/server/srv/tftp/<MyRPI_SN>/config.txt to 192.168.0.25
May 28 12:48:17 dnsmasq-tftp[676]: file /media/server/srv/tftp/<MyRPI_SN>/recovery.elf not found
May 28 12:48:20 dnsmasq-tftp[676]: file /media/server/srv/tftp/<MyRPI_SN>/fixup.dat not found
May 28 12:48:20 dnsmasq-tftp[676]: sent /media/server/srv/tftp/<MyRPI_SN>/start.elf to 192.168.0.25
May 28 12:48:20 dnsmasq-tftp[676]: file /media/server/srv/tftp/<MyRPI_SN>/recovery.elf not found
May 28 12:48:20 dnsmasq-tftp[676]: sent /media/server/srv/tftp/<MyRPI_SN>/config.txt to 192.168.0.25
May 28 12:48:20 dnsmasq-tftp[676]: file /media/server/srv/tftp/<MyRPI_SN>/dt-blob.bin not found
May 28 12:48:20 dnsmasq-tftp[676]: file /media/server/srv/tftp/<MyRPI_SN>/recovery.elf not found
May 28 12:48:20 dnsmasq-tftp[676]: sent /media/server/srv/tftp/<MyRPI_SN>/config.txt to 192.168.0.25
May 28 12:48:20 dnsmasq-tftp[676]: file /media/server/srv/tftp/<MyRPI_SN>/bootcfg.txt not found
May 28 12:48:20 dnsmasq-tftp[676]: sent /media/server/srv/tftp/<MyRPI_SN>/cmdline.txt to 192.168.0.25
May 28 12:48:20 dnsmasq-tftp[676]: file /media/server/srv/tftp/<MyRPI_SN>/recovery8.img not found
May 28 12:48:20 dnsmasq-tftp[676]: file /media/server/srv/tftp/<MyRPI_SN>/recovery8-32.img not found
May 28 12:48:20 dnsmasq-tftp[676]: file /media/server/srv/tftp/<MyRPI_SN>/recovery7.img not found
May 28 12:48:20 dnsmasq-tftp[676]: file /media/server/srv/tftp/<MyRPI_SN>/recovery.img not found
May 28 12:48:20 dnsmasq-tftp[676]: file /media/server/srv/tftp/<MyRPI_SN>/kernel8.img not found
May 28 12:48:20 dnsmasq-tftp[676]: file /media/server/srv/tftp/<MyRPI_SN>/kernel8-32.img not found
May 28 12:48:20 dnsmasq-tftp[676]: file /media/server/srv/tftp/<MyRPI_SN>/kernel7.img not found
May 28 12:48:20 dnsmasq-tftp[676]: file /media/server/srv/tftp/<MyRPI_SN>/kernel.img not found
May 28 12:48:20 dnsmasq-tftp[676]: file /media/server/srv/tftp/<MyRPI_SN>/armstub8.bin not found
May 28 12:48:20 dnsmasq-tftp[676]: file /media/server/srv/tftp/<MyRPI_SN>/armstub8-32.bin not found
May 28 12:48:20 dnsmasq-tftp[676]: file /media/server/srv/tftp/<MyRPI_SN>/armstub7.bin not found
May 28 12:48:20 dnsmasq-tftp[676]: file /media/server/srv/tftp/<MyRPI_SN>/armstub.bin not found
May 28 12:48:20 dnsmasq-tftp[676]: file /media/server/srv/tftp/<MyRPI_SN>/bcm2710-rpi-3-b.dtb not found
```
Baisically it seems `<MyRPI_SN>/` should have much more than just `cmdline.txt config.txt rpi-raspbian-lite.url ssh`. Is this my fault or something going on with the install script?

HarvsG, 2018-05-28:
Seems that this line isn't doing anything on my system:
https://github.com/beta-tester/RPi-PXE-Server/blob/383b85432b3b18ac43de0eb36f210eb4d1d1e229/install-pxe-server_pass2.sh#L1334

I think it might actually be failing here: https://github.com/beta-tester/RPi-PXE-Server/blob/383b85432b3b18ac43de0eb36f210eb4d1d1e229/install-pxe-server_pass2.sh#L1228

Looking closely at the install logs I see 
```
Archive:  /media/server/srv/img/raspbian_lite_latest.zip
  inflating: /media/server/srv/img/2018-04-18-raspbian-stretch-lite.img
mv: cannot stat '/media/server/srv/img/raspbian_lite_latest.img': No such file or directory
touch: failed to get attributes of '/media/server/srv/img/rpi-raspbian-lite.img': No such file or directory
```
Seems like this line is not working well: https://github.com/beta-tester/RPi-PXE-Server/blob/383b85432b3b18ac43de0eb36f210eb4d1d1e229/install-pxe-server_pass2.sh#L1177
Indeed `echo "$RAW_FILENAME_IMG";`
returned: `raspbian_lie_latest.img`
So it seems that it is not actually getting the raw file name

HarvsG, 2018-05-28:
I have found the problem!
https://github.com/beta-tester/RPi-PXE-Server/blob/383b85432b3b18ac43de0eb36f210eb4d1d1e229/install-pxe-server_pass2.sh#L215
The url redirects to a name in the format `2018-04-18-raspbian-stretch-lite.zip`

But when `basename()` runs on the above urls it returns `'raspbian_lite_latest'` and not `2018-04-18-raspbian-stretch-lite`:https://github.com/beta-tester/RPi-PXE-Server/blob/383b85432b3b18ac43de0eb36f210eb4d1d1e229/install-pxe-server_pass2.sh#L1176-L1177

Which means this `mv` fails https://github.com/beta-tester/RPi-PXE-Server/blob/383b85432b3b18ac43de0eb36f210eb4d1d1e229/install-pxe-server_pass2.sh#L1228

beta-tester, 2018-05-28:
oh yeah... gotcha...
i forgot to copy that part from branch testing to branch master.
hope it will work now correctly.

thank you for reporting.

closed 2018-05-29 by beta-tester