beta-tester/RPi-PXE-Server #9: [Not boot] The PXE function is not work on my Raspberry Pi3

Almoon, 2018-06-15:
Hi sir, 
First of all, thanks for providing this great project.
The RPi PXE server does not work on my side.
I tried to install your default shell 1 and 2 files. (Download some ISO files on my 64GB card)
I also tried to modified some changes on the pass_2.sh (add/remove ISO path, change “DST_ROOT=/srv “or “/DST_ROOT=/srv to DST_ROOT=/media/server/srv”)

Did I miss something?

I reinstalled my system and failed over 20 times, so sad ....
Attaching three log files on this comment.
Thanks.

------------- System equipment ----------
a. RPi3 mode B
- SD card: 16GB, 32GB, 64GB
- USB device: SanDisk 32GB
- OS: NOOBS_v2_8_1
b. Client: Enable PXE IPv4/6 features in BIOS.

Wired LAN schematic:
1.	RPi3 – client
2.	RPi3- Hub - client
--------------------------------------------------
The Steps:
0. Plug a 64GB SD card on the RPi3 Model B.
1. Plug a USB 32GB memory device on the RPi3.
2. Power on and then boot to the desktop.
3. $ sudo apt-get update
4. $ sudo apt-get upgrade
5. Umount my USB 32GB device. (sda1)
5. $ sudo mkfs.ext4 -L PXE-Server /dev/sda1 (sda1 is my 32GB USB device, LABEL: PXE- Server)
6. Copy the install shell files on the desktop. 
7. $ bash install-pxe-server_pass1.sh
8. reboot the system
-9. pi@raspberrypi:/media/server $ sudo mkdir -p ./backup/img
-10.pi@raspberrypi:/media/server/backup $ sudo mkdir -p ./backup/iso
-11.pi@raspberrypi:/media/server/backup $ sudo mkdir -p ./backup/tftp/boot
-12.pi@raspberrypi:/media/server/backup $ sudo mkdir -p ./backup/tftp/boot
13.pi@raspberrypi:~/Desktop/RPi-PXE-Server-master $ bash install-pxe-server_pass2.sh
14. Reboot to RPi desktop.
15. Done.
16. The “PXE-E51: No DHCP or proxyDHCP offers were received.” Message is displayed on the client system.

[check_service_status.txt](https://github.com/beta-tester/RPi-PXE-Server/files/2105769/check_service_status.txt)
[install1file_log.txt](https://github.com/beta-tester/RPi-PXE-Server/files/2105770/install1file_log.txt)
[install2file_log.txt](https://github.com/beta-tester/RPi-PXE-Server/files/2105771/install2file_log.txt)

beta-tester, 2018-06-15:
thank you for creating and providing the log files...

> -------- is used as SN for RPi3 network booting

> handle_network_booting(rpi-raspbian-lite)
>     skipped: no serial number setted at RPI_SN0.

i looks like you forgot to add the eight last digits of your RPi's serial number to the script 2.
see [note4: NETWORK BOOTING for Raspberry Pi 3](https://github.com/beta-tester/RPi-PXE-Server#note4-network-booting-for-raspberry-pi-3)

Almoon, 2018-06-19:
Hi beta-tester,
Thanks for your prompt feedback.
My RPi PXE server works after I added the serial number to the script2 file.

I did some tests below:
- Legacy IPv4 : Pass
- UEFI IPv4: Fail
- UEFI IPv6: Fail
I uploaded some pictures for you.

[image]

[image]

---- System equipment ---
a. RPi3 mode B : 2 boards
- SD card: 32GB, 64GB
- USB device: SanDisk 32GB
- OS: NOOBS_v2_8_1
b. Client: Enable PXE IPv4/6 features in BIOS.
- Run a system in the VirtualBox (legacy IPv4)
- Two systems supported PXE boot (legacy IPv4, UEFI IPv4/6)

Today, I realized that I made two failures on this project:
1. I misunderstood that RPi's serial number before. I thought that the 8 numbers was for a client RPi device.
2. When the RPi-PXE server updated success, I unplugged my WLAN from the Hub.
So after reboot or power on my RPi-PXE server that it unable to got an IP. (Because it has a risk in a public/office network, I don’t want to make a network storm.)
Fortunately, it works on my office network (there is an official DHCP server in my office.)

Can I run the RPi-PXE server in an offline mode? (like: PXE- hub- Client, or PXE- Client )
If you need me to do other tests, please let me know.
- Thanks.

beta-tester, 2018-06-19:
if the RPi-PXE-Server has a fix IP (that is sould when a virgin RPi-Raspbian-Stretch image was used and an unmodified script), then it sould be possible to run that RPi-PXE-Server in an isolated network with a switch and its clients only.

the issue with UEFI, is it related to RPi3-pxe boot or is it related to PC-pxe-boot?
if it is related to PC-pxe-boot, you may add your issue to the existing issue #6 and reopen it.
i am not able to test UEFI networkboot, because my hardware offers my only BIOS-Legacy-PXE-Boot. UEFI IPv4/IPv6 is not available to me.
i thought with issue #6 it is fixed.

closed 2018-06-19 by beta-tester

Almoon, 2018-06-20:
It related to PC-pxe-boot with UEFI mode.
I will not reopen the issue #6 until i have a breakthrough on UEFI pxe boot.

Thanks for your great support.