beta-tester/RPi-PXE-Server #10: Troubleshooting

bradbrundage, 2018-12-18:
I have everything installed from the readme (ISO's are downloaded as well via the scripts), but am having an issue that I haven't figured out yet.

I see the menu but when I select an item to boot from it says it's downloading, however never does anything and then returns to the PXE boot screen. Any ideas?

Running latest raspbian Linux raspberrypi 4.14.79-v7+ fully updated attaching my installed packages in case there's a version that needs to be different than installed.

[Installed_packages.txt](https://github.com/beta-tester/RPi-PXE-Server/files/2691679/Installed_packages.txt)

beta-tester, 2018-12-18:
hi,

when it returns just to the PXE menu, then the bootprocess could not load the kernel or the initrd file.
you can try to remove the `http://pxe-server` in the `/srv/tftp/menu-bios/pxelinux.cfg/default`
e.g. 
```
    KERNEL http://pxe-server/nfs/ubuntu-x64/casper/vmlinuz
    INITRD http://pxe-server/nfs/ubuntu-x64/casper/initrd
```
to
```
    KERNEL /nfs/ubuntu-x64/casper/vmlinuz
    INITRD /nfs/ubuntu-x64/casper/initrd
```
what does `tree -L 2 -d /srv/tftp` show?
sould be somethin like:
```
pi@pxe-server:~ $ tree -L 2 -d /srv/tftp
/srv/tftp
├── menu-bios
│   ├── iso -> /srv/iso/
│   ├── nfs -> /srv/nfs/
│   └── pxelinux.cfg

```

did you modified the script?
which iso's did you downloaded and tried out to boot?
which PXE boot menu do you use - "legacy" BIOS or UEFI ?
UEFI is not tested well

bradbrundage, 2018-12-18:
Thank you for the reply, I did determine what was causing the issue. In /etc/hosts the default hosts file defines the hostname raspberrypi to it's loopback address 127.0.0.1 once I changed that to the IP of the pi and rebooted it will now load the iso image I was trying to boot.

tailing the logs I notice this but am guessing it's normal based on what BIOS/UEFI may be in use on a machine that's booting. The machine booting is an old Core2Duo Dell so as far as I see in the BIOS that it doesn't support UEFI.

raspberrypi dnsmasq-tftp[597]: error 0 TFTP Aborted received from 192.168.1.232
raspberrypi dnsmasq-tftp[597]: failed sending /srv/tftp/menu-bios/lpxelinux.0 to 192.168.1.232

I'm mainly trying to get CloneZilla, WinPE, DBAN (will try to figure out how to add) and the kasperski offline scanner (I updated the URL and it downloaded fine)

Here's the results from tree:
root@raspberrypi:~# tree -L 2 -d /srv/tftp
/srv/tftp
├── boot
├── menu-bios
│   ├── iso -> /srv/iso/
│   ├── nfs -> /srv/nfs/
│   └── pxelinux.cfg
├── menu-efi32
│   ├── iso -> /srv/iso/
│   ├── nfs -> /srv/nfs/
│   └── pxelinux.cfg
├── menu-efi64
│   ├── iso -> /srv/iso/
│   ├── nfs -> /srv/nfs/
│   └── pxelinux.cfg
└── sources -> /srv/nfs/win-pe-x86/sources/

bradbrundage, 2018-12-18:
I tried enabling DESINFECT_X64 by adding the iso URL and added the handle but it never shows up. Also trying to add DBAN and it shows on the menu but does not load. Attached is the script that I modified.

Thank you!

[install-pxe-server_pass2.txt](https://github.com/beta-tester/RPi-PXE-Server/files/2692431/install-pxe-server_pass2.txt)

beta-tester, 2018-12-19:
it looks to me, you swapped the IF-condition of the MENU-entry of clonecilla.

```
    if [ -f "$FILE_MENU" ] \
    && [ -f "$DST_NFS_ETH0/$CLONEZILLA_X64/live/vmlinuz" ]; then
        echo  -e "\e[36m    add $CLONEZILLA_X64\e[0m";
        sudo sh -c "cat << EOF  >> $FILE_MENU
########################################
LABEL $DBAN_X86
    MENU LABEL DBAN x86
    KERNEL $FILE_BASE$NFS_ETH0/$DBAN_X86/DBAN.BZI
    APPEND nfsmount=$IP_ETH0:$DST_NFS_ETH0/$DBAN_X86 ro netboot=nfs
    TEXT HELP
        Boot to DBAN x86
    ENDTEXT
EOF";
    fi

    if [ -f "$FILE_MENU" ] \
    && [ -f "$DST_NFS_ETH0/$DBAN_X86/DBAN.BZI" ]; then
       echo  -e "\e[36m    add $DBAN_X86\e[0m";
       sudo sh -c "cat << EOF  >> $FILE_MENU
########################################
LABEL $CLONEZILLA_X64
    MENU LABEL Clonezilla x64
    KERNEL $FILE_BASE$NFS_ETH0/$CLONEZILLA_X64/live/vmlinuz
    INITRD $FILE_BASE$NFS_ETH0/$CLONEZILLA_X64/live/initrd.img
    APPEND nfsroot=$IP_ETH0:$DST_NFS_ETH0/$CLONEZILLA_X64 ro netboot=nfs boot=live config username=user hostname=clonezilla union=overlay components noswap edd=on nomodeset nodmraid ocs_live_run=ocs-live-general ocs_live_extra_param= ocs_live_batch=no net.ifnames=0 nosplash noprompt -- locales=$CUSTOM_LANG_LONG.UTF-8 keyboard-layouts=$CUSTOM_LANG utc=no timezone=$CUSTOM_TIMEZONE
    TEXT HELP
        Boot to Clonezilla x64
        User: user, Password: live
    ENDTEXT
EOF";
    fi
```
try this...
```
    if [ -f "$FILE_MENU" ] \
    && [ -f "$DST_NFS_ETH0/$DBAN_X86/DBAN.BZI" ]; then
       echo  -e "\e[36m    add $DBAN_X86\e[0m";
       sudo sh -c "cat << EOF  >> $FILE_MENU
########################################
LABEL $DBAN_X86
    MENU LABEL DBAN x86
    KERNEL $FILE_BASE$NFS_ETH0/$DBAN_X86/DBAN.BZI
    APPEND nfsmount=$IP_ETH0:$DST_NFS_ETH0/$DBAN_X86 ro netboot=nfs
    TEXT HELP
        Boot to DBAN x86
    ENDTEXT
EOF";
    fi
```

BTW: when you hit [TAB]-key in the PXE boot menu, you can see and edit the boot parameter just before your kernel loads.

beta-tester, 2018-12-19:
are the names of the mounted filesystem of DBAN really all with uppercase letters?
`    KERNEL $FILE_BASE$NFS_ETH0/$DBAN_X86/DBAN.BZI`
`KERNEL $FILE_BASE$NFS_ETH0/$DBAN_X86/dban.biz`

beta-tester, 2018-12-19:
oh, and be careful with the APPEND parameters you give,
because i saw, that it is possible to automaticaly wipe all data on the computer who boots DBAN.

bradbrundage, 2018-12-19:
> oh, and be careful with the APPEND parameters you give,
> because i saw, that it is possible to automaticaly wipe all data on the computer who boots DBAN.
> see isolinux.cfg of DBAN to get a clue of what parameters are possible.

Correct, I was planning on just booting to the GIU where you can select the wipe parameters.

bradbrundage, 2018-12-19:
> ########################################

So "########################################" is not separating the menu items?

I thought it supposed to look like this:
```
######################################## <Beginning of ITEM1>
LABEL $ITEM1
    MENU LABEL ITEM1
    KERNEL $FILE_BASE$NFS_ETH0/$ITEM1/kernelFile.ext
    APPEND nfsmount=$IP_ETH0:$DST_NFS_ETH0/$ITEM1 ro netboot=nfs
    TEXT HELP
        Boot to ITEM1 x86
    ENDTEXT
EOF";
    fi

    if [ -f "$FILE_MENU" ] \
    && [ -f "$DST_NFS_ETH0/$ITEM1/kernelFile.ext" ]; then
       echo  -e "\e[36m    add $ITEM1\e[0m";
       sudo sh -c "cat << EOF  >> $FILE_MENU
######################################## <End of ITEM1>
```
 But it appears that the menu items before the "########################################" that I thought were from the previous block are to be part of the block for that item but instead should be referencing the item before?

beta-tester, 2018-12-19:
opening the iso under windows with 7-Zip may display all files in uppercase...
but you have to look how it is mounted under linux.
in your case take a look to `/srv/nfs/DBAN-X86/`...

`########################################` is not neant as separator for the script. my intention was to separate the entries in the generated /srv/tftp/menu-bios/pxelinux.cfg/default pxe boot menu.
maybe i should it make more clear in the script as well...

bradbrundage, 2018-12-21:
> > are the names of the mounted filesystem of DBAN really all with uppercase letters?
> > ` KERNEL $FILE_BASE$NFS_ETH0/$DBAN_X86/DBAN.BZI`
> > `KERNEL $FILE_BASE$NFS_ETH0/$DBAN_X86/dban.bzi`
> 
> When I open the iso file with 7-Zip it is in all uppercase.

That was the issue with the file name. Now I have an issue with Desinfect which is now KRD downloaded from:
http://rescuedisk.kaspersky-labs.com/rescuedisk/updatable/2018/krd.iso

Here's the part of the config, I can get the 64 bit version to boot (but does not function correctly) and the 32 bit never shows up on the menu. any suggestions?

```
########################################
LABEL $DESINFECT_X86
    MENU LABEL desinfect x86
    KERNEL $FILE_BASE$NFS_ETH0/$DESINFECT_X86/boot/grub/k-x86
    INITRD $FILE_BASE$NFS_ETH0/$DESINFECT_X86/boot/grub/initrd.xz
    APPEND nfsroot=$IP_ETH0:$DST_NFS_ETH0/$DESINFECT_X86 ro netboot=nfs file=/cdrom/preseed/ubuntu.seed boot=casper memtest=4 rmdns -- debian-installer/language=$CUSTOM_LANG console-setup/layoutcode=$CUSTOM_LANG keyboard-configuration/layoutcode=$CUSTOM_LANG keyboard-configuration/variant=$CUSTOM_LANG_WRITTEN
    TEXT HELP
        Boot to ct desinfect x86
        User: desinfect
    ENDTEXT
EOF";
    fi

    if [ -f "$FILE_MENU" ] \
    && [ -f "$DST_NFS_ETH0/$DESINFECT_X86/boot/grub/k-x86" ]; then
        echo  -e "\e[36m    add $DESINFECT_X86\e[0m";
        sudo sh -c "cat << EOF  >> $FILE_MENU
########################################
LABEL $DESINFECT_X64
    MENU LABEL desinfect x64
    KERNEL $FILE_BASE$NFS_ETH0/$DESINFECT_X64/boot/grub/k-x86_64
    INITRD $FILE_BASE$NFS_ETH0/$DESINFECT_X64/boot/grub/initrd.xz
    APPEND nfsroot=$IP_ETH0:$DST_NFS_ETH0/$DESINFECT_X64 ro netboot=nfs file=/cdrom/preseed/ubuntu.seed boot=casper memtest=4 rmdns -- debian-installer/language=$CUSTOM_LANG console-setup/layoutcode=$CUSTOM_LANG keyboard-configuration/layoutcode=$CUSTOM_LANG keyboard-configuration/variant=$CUSTOM_LANG_WRITTEN
    TEXT HELP
        Boot to ct desinfect x64
        User: desinfect
    ENDTEXT
EOF";
    fi

    if [ -f "$FILE_MENU" ] \
    && [ -f "$DST_NFS_ETH0/$DESINFECT_X64/boot/grub/k-x86_64" ]; then
        echo  -e "\e[36m    add $DESINFECT_X64\e[0m";
        sudo sh -c "cat << EOF  >> $FILE_MENU
########################################

```

beta-tester, 2018-12-21:
1'st: and again, you mixed up the IF-condition and MENU-content
you should update your clone of my project files. i added `#--- BEGIN ---` and `#---- END ----` to the script to show, where a block starts and where it ends.

2'nd: the initrd of the kaspersky iso seams not to been made for PXE boot.
you have to adjust the init-script inside the initrd.gz
or overlay initrd.gz with a customized one.
try this:
```
KASPERSKY_RESCUE_X86=kaspersky-rescue-x86
KASPERSKY_RESCUE_X86_URL=https://rescuedisk.s.kaspersky-labs.com/updatable/2018/krd.iso
```
and
```
#========== BEGIN ==========
    if [ -f "$FILE_MENU" ] \
    && [ -f "$DST_NFS_ETH0/$KASPERSKY_RESCUE_X86/boot/grub/k-x86_64" ]; then
        echo  -e "\e[36m    add $KASPERSKY_RESCUE_X86\e[0m";
        [ -f "$DST_NFS_ETH0/INITRD_N24.1.GZ" ] || sudo wget -O $DST_NFS_ETH0/INITRD_N24.1.GZ https://www.vercot.com/~serva/download/INITRD_N24.1.GZ
        sudo sh -c "cat << EOF  >> $FILE_MENU
########################################
LABEL $KASPERSKY_RESCUE_X86
    MENU LABEL Kaspersky Rescue CD x86
    KERNEL $FILE_BASE$NFS_ETH0/$KASPERSKY_RESCUE_X86/boot/grub/k-x86_64
    INITRD $FILE_BASE$NFS_ETH0/$KASPERSKY_RESCUE_X86/boot/grub/initrd.xz,$FILE_BASE$NFS_ETH0/INITRD_N24.1.GZ
    APPEND netboot=$FILE_BASE$NFS_ETH0/$KASPERSKY_RESCUE_X86/data/* ro loadsrm=000-core.srm,001-xorg.srm,002-xfce.srm,003-kl.srm,004-krt.srm,005-bases.srm,008-firefox.srm net.ifnames=0 dodhcp dostartx
    TEXT HELP
        Boot to Kaspersky Rescue CD 32bit
    ENDTEXT
EOF";
    fi
#=========== END ===========
```
and
```
handle_iso  $KASPERSKY_RESCUE_X86  $KASPERSKY_RESCUE_X86_URL  timestamping;
```

WARNING: downloading and using the external custom initrd file may violate the copyright of its original owner.

beta-tester, 2018-12-22:
to speed up boot, you can change the append to:
```
    APPEND netboot=nfs://$IP_ETH0:$DST_NFS_ETH0/$KASPERSKY_RESCUE_X86 ro dostartx
```

in fact, what is wrong in the original init-script is that for unknown reason the checksum md5 is in the script, but sha512 file are in the iso. and the /etc/resolv.conf will not be copied correctly.

here the patch:
```
--- init-original	2018-05-23 14:59:15.000000000 +0200
+++ init	2018-12-22 09:09:04.863630241 +0100
@@ -493,7 +493,7 @@
                 echo "--- ${cureth}: link=${linkstate}, mac=${mac}"
 
                 # If the link is up, then try (even if another interface already has a dhcp address)
-                if [ "${linkstate}" != 'no-link' ] 
+                if [ "${linkstate}" != 'no-link' ]
                 then
                     good_msg "Attempting to get a DHCP address on ${cureth}..."
                     [ -n "${DHCPHOSTNAME}" ] && hostnameopt="-H ${DHCPHOSTNAME}"
@@ -639,7 +639,7 @@
     for curfile in ${FILENAMES}
     do
         filename="${curfile}"
-        filecksm=$(echo ${curfile} | sed -e 's!.dat!.md5!' -e 's!.srm!.md5!') 
+        filecksm=$(echo ${curfile} | sed -e 's!.dat!.sha512!' -e 's!.srm!.sha512!')
         urldatafile="${BASEADDR}/${filename}"
         urlchecksum="${BASEADDR}/${filecksm}"
         good_msg "Downloading ${urldatafile} ..."
@@ -665,13 +665,13 @@
             sysresccd_panic "Cannot download the \"${filename}\" boot file."
         fi
 
-        md5now=$(/bin/busybox md5sum ${BOOTPATH}/${filename} | cut -d ' ' -f1)
-        md5exp=$(cat ${BOOTPATH}/${filecksm} | cut -d ' ' -f1)
-        if [ "${md5now}" = "${md5exp}" ]
+        sha512now=$(/bin/busybox sha512sum ${BOOTPATH}/${filename} | cut -d ' ' -f1)
+        sha512exp=$(cat ${BOOTPATH}/${filecksm} | cut -d ' ' -f1)
+        if [ "${sha512now}" = "${sha512exp}" ]
         then
-            good_msg "Successfully downloaded and md5 checked \"${filename}\""
+            good_msg "Successfully downloaded and sha512 checked \"${filename}\""
         else
-            sysresccd_panic "Filesystem image ${filename} is corrupt: md5sum incorrect"
+            sysresccd_panic "Filesystem image ${filename} is corrupt: sha512sum incorrect"
         fi
     done
 }
@@ -710,7 +710,7 @@
     for curfile in ${FILENAMES}
     do
         filename="${curfile}"
-        filecksm=$(echo ${curfile} | sed -e 's!.dat!.md5!' -e 's!.srm!.md5!') 
+        filecksm=$(echo ${curfile} | sed -e 's!.dat!.sha512!' -e 's!.srm!.sha512!')
         urldatafile="${tftpurl}/${filename}"
         urlchecksum="${tftpurl}/${filecksm}"
         good_msg "Downloading ${urldatafile} ..."
@@ -730,13 +730,13 @@
             sysresccd_panic "Cannot download the files: tftp failed. May be due to lack of memory"
         fi
 
-        md5now=$(/bin/busybox md5sum ${BOOTPATH}/${filename} | cut -d ' ' -f1)
-        md5exp=$(cat ${BOOTPATH}/${filecksm} | cut -d ' ' -f1)
-        if [ "${md5now}" = "${md5exp}" ]
+        sha512now=$(/bin/busybox sha512sum ${BOOTPATH}/${filename} | cut -d ' ' -f1)
+        sha512exp=$(cat ${BOOTPATH}/${filecksm} | cut -d ' ' -f1)
+        if [ "${sha512now}" = "${sha512exp}" ]
         then
-            good_msg "Successfully downloaded and md5 checked \"${filename}\""
+            good_msg "Successfully downloaded and sha512 checked \"${filename}\""
[... 84 lines cut ...]
+    cp /etc/resolv.conf ${NEWROOT}/etc/
 }
 
 sysresccd_stage3_rootsys() # mount the root partition on ${SQUASHFSMNT}
@@ -1757,7 +1759,7 @@
 BACKSTORE_MEM='/memory'
 BACKSTORE_DAT='sysrcd.bs'
 LOOPDAT='kernel.dat'
-LOOPMD5='kernel.dat.md5'
+LOOPSHA512='kernel.dat.sha512'
 INITPROG='/sbin/init'
 CONSOLE='/dev/console'
 CACHEDIR='/cache'
```
to create the new patched init script you can create your custom initrd.xz file you can overlay
```
sudo chmod 555 init
find . | cpio --create --format=newc > /tmp/kaspersky-rescue-x86-initrd
xz --check=crc32 /tmp/kaspersky-rescue-x86-initrd
sudo mv /tmp/kaspersky-rescue-x86-initrd.xz /srv/nfs/.
```
then you can use the ner initrd:
```
    INITRD $FILE_BASE$NFS_ETH0/$KASPERSKY_RESCUE_X86/boot/grub/initrd.xz,$FILE_BASE$NFS_ETH0/kaspersky-rescue-x86-initrd.xz
```

beta-tester, 2018-12-22:
see also [issues: PXE-boot free Kaspersky Rescue Disk](https://forum.kaspersky.com/index.php?/topic/404641-issues-pxe-boot-free-kaspersky-rescue-disk/)

beta-tester, 2018-12-23:
i just added free kaspersky rescue disk to the project.
see commit  https://github.com/beta-tester/RPi-PXE-Server/commit/c38c8f3f33e06e691b9da6e1ac424b957e9d3372

closed 2018-12-23 by beta-tester

bradbrundage, 2018-12-24:
> i just added free kaspersky rescue disk to the project.
> see commit [c38c8f3](https://github.com/beta-tester/RPi-PXE-Server/commit/c38c8f3f33e06e691b9da6e1ac424b957e9d3372)

Thank you, I'll check everything out after the holidays.

beta-tester, 2018-12-24:
please note, i splitted the install-pxe-server-pass2.sh script in to multiple scripts.
p2-include-var.sh (includes variables like directory names and ip adresses),
p2-include-url.sh (includes the url and names of the images),
p2-include-menu.sh (includes the pxe manu enties with the kernel parameters),
p2-include-handle.sh (includes the handles that controls what image will be exposed by the pxe-server).

you still only have to execute the install-pxe-server-pass2.sh only.
this script will execute all the p2-include-... scripts.

i hope it is easier to maintain modification.
instead of searching one big script to find the interesting parts to modify, you now will find it in the separate files.