beta-tester/RPi-PXE-Server #11: Patch

andrecbarros, 2019-01-02:
Can you post here the patch you applied to kaspersky rescue disk?

beta-tester, 2019-01-02:
there the patch as attachement:
[init.patch.zip](https://github.com/beta-tester/RPi-PXE-Server/files/2721289/init.patch.zip)

you can find the patch also on this thread: https://github.com/beta-tester/RPi-PXE-Server/issues/10#issuecomment-449555361
(there is also an patched initrd.xz)

or https://forum.kaspersky.com/index.php?/topic/397563-krd-2018-pxe/&tab=comments#comment-2843816

andrecbarros, 2019-01-02:
No relation with Kaspersky, just like their antivirus product and have to administer some PXE servers for backup services and emergency happenings.

Thanks for sharing!

closed 2019-01-03 by andrecbarros

beta-tester, 2019-09-07:
it was now pathed by kaspersky...
should now work without the extra init file.
see https://community.kaspersky.com/kaspersky-rescue-disk-78/please-fix-init-file-of-the-rescure-dvd-for-proper-pxe-boot-970