beta-tester/RPi-PXE-Server #12: Broken already setted up Samba shares

Andriy2, 2019-01-29:
After installing PXE Server my setted up sama shares was lost!

beta-tester, 2019-01-30:
sorry for that...
yes, at the moment i assumed a new fresh virgin raspbian installation.
so i cut away everything, that is comming after the
`#======================= Share Definitions =======================` comment in the /etc/samba/smb.conf file and add shares  for /srv and /media...
then better i will keep existing settings and simply add my stuff.

beta-tester, 2019-01-30:
i just changed that behavior in the master branch...
but there are more config files that will be simply replaces instead of reconfigured for simplicity.
even more in the testing branch - there i don't take care for anything - that is my personal testing branch.

i added some addition hints to the readme file, that things will be overridden.

closed 2019-01-30 by beta-tester