beta-tester/RPi-PXE-Server #13: How to use $src_iso and $file_iso

Bigjohn4, 2019-03-06:
I am trying to get any Linux running on old laptop. Laptop will not boot DVD or USB drive. your PXE server worked well for many of the built in URLs and also several I added. But, several I want to try will not download, although I can download them from Firefox. They have lots of redirects, which could be the issue.
I looking at pass2.sh, I found $src_iso and $file_iso. I put the ISOs I want to use on memory stick, but have been unable to get it to work. Do you have documentation on using these options?
[using file name.txt](https://github.com/beta-tester/RPi-PXE-Server/files/2938545/using.file.name.txt)

 I tried above file, but it is not readable. Do I need a translation? 
I am including my configs and screen capture for my last try. I am using Raspberry Pi for PXE server.

beta-tester, 2019-03-07:
hello,
i am not able to open/see the linked content of "using file name.txt"... it is a zip file with no useful content.
this file isn't from me. maybe an attachement from somebody who wrote an issue or so.
from where did you got that link?

what ISO did you tried? then i maybe can make an example.

to download, mount and export an ISO, you have to add:
to p2-import-url.sh:
```
MY_ISO=<uniqe_name_without_blanks>
MY_ISO_URL=<url_to_iso_file>

```
to p2-include-handle.sh:
```
handle_iso $MY_ISO $MY_ISO_URL;

```

and to get it listed in the menu, add to p2-include-menu.sh:
```
#========== BEGIN ==========
if [ -f "$FILE_MENU" ] \
&& [ -f "$DST_NFS_ETH0/$MY_ISO/<path to kernel file e.g. vmlinuz>" ]; then
    echo  -e "\e[36m    add $MY_ISO\e[0m";
    sudo sh -c "cat << EOF  >> $FILE_MENU
    ########################################
    LABEL $MY_ISO
        MENU LABEL <name of iso live system>
        KERNEL $FILE_BASE$NFS_ETH0/$MY_ISO/<path to kernel file e.g. vmlinuz>
        INITRD $FILE_BASE$NFS_ETH0/$MY_ISO/<path to initrd file, e.g. initrd.img>
        APPEND nfsroot=$IP_ETH0:$DST_NFS_ETH0/$MY_ISO ro <boot options of your iso live system>
EOF";
fi
#=========== END ===========

```

you have to replace all `MY_ISO` & `MY_ISO_URL` to your preffered variable name.
and everything in between `<...>` to the correct what ever is needed.

PS.: if you don't have an earlier version of RPi-PXE-Server cloned, then you don't have the stuff in seperated files. then you have to put those stuff in the `install-pxe-server_pass2.sh` file to the right spots/places.

for the kernel and initrd file locations/names, you have to look into the iso to see where they are and what name they have.

for boot options you have to read the documentation of the used system and or watch into severan .cfg file in the iso.
when the system is based of an other system that already is included in the project, you maybe can recyckle those boot options.

Bigjohn4, 2019-03-07:
When I was typing the issue, the text file name popped up.
Thanks, I'll give the above a try and let you know. below is one that would not fetch, although I got the URL to work in Netscape.
LINUX_LITE_URL=http://pumath.dl.osdn.jp/linux-lite-3.0-32bit.iso

closed 2019-03-07 by Bigjohn4

beta-tester, 2019-03-07:
in firefox i get only 404 Not Found.

Bigjohn4, 2019-03-10:
I downloaded the zip of latest scripts, and started with just one ISO.
Still no luck. What am I missing?
Attached file has changes I made to scripts.

JM

p2-include-handle.sh
#===== jwm =====
handle_iso  $LINUX_LITE $LINUX_LITE_URL $LINUX_LITE_SRC_ISO $LINUX_LITE_FILE_ISO;
#handle_iso  $LINUX_LITE $LINUX_LITE_URL;
#===== jwm =====
p2-include-menu.sh
#===== jwm =====
#========== BEGIN ==========
if [ -f "$FILE_MENU" ] \
&& [ -f "$DST_NFS_ETH0/$LUNUX_LITE/casper/vmlinuz" ]; then
    echo  -e "\e[36m    add $LINUX_LITE\e[0m";
    sudo sh -c "cat << EOF  >> $FILE_MENU
    ########################################
    LABEL $LINUX_LITE
        MENU LABEL LINUX LITE
        KERNEL $FILE_BASE$NFS_ETH0/$LINUX_LITE/casper/vmlinuz
        INITRD $FILE_BASE$NFS_ETH0/$LINUX_LITE/casper/initrd.gz
        APPEND nfsroot=$IP_ETH0:$DST_NFS_ETH0/$LINUX_LITE ro netboot=nfs boot=live config  username=linux --
 locales=$CUSTOM_LANG_LONG.UTF-8 keyboard-layouts=$CUSTOM_LANG utc=no timezone=$CUSTOM_TIMEZONE
        TEXT HELP
            Boot to Linux Lite
            User:linux , Password:
        ENDTEXT
EOF";
fi
#=========== END ===========
#===== jwm =====
p2-include-url.sh
#=== jwm ===
#
LINUX_LITE=linux_lite
#LINUX_LITE_URL=http://pumath.dl.osdn.jp/linux-lite-3.0-32bit.iso
LINUX_LITE_URL=
LINUX_LITE_SRC_ISO=/media/pi/3032-6134         <<== tried both ways
#LINUX_LITE_SRC_ISO=/media/pi/3032-6134/      <<==
LINUX_LITE_FILE_ISO=linux-lite-3.8-32bit.iso
#
#=== jwm ===
p2-include-var.sh
#### jwm ####
CUSTOM_LANG=en
CUSTOM_LANG_LONG=en_EN
CUSTOM_LANG_UPPER=EN
CUSTOM_LANG_WRITTEN=English
CUSTOM_LANG_EXT=us-english-nodeadkeys
CUSTOM_TIMEZONE=US/central
RPI_SN0=81c915c7
INTERFACE_ETH0=eth0
#### jwm ####
-------------------------------------------------------------------
/dev/sda1 on /media/pi/3032-6134 type vfat .....
pi@raspberrypi:~/PXE $ ls -l /media/pi/3032-6134
total 2626080
-rw-r--r-- 1 pi pi     138145 Mar  8 14:39 bhodi.iso
-rw-r--r-- 1 pi pi          0 Mar  8 14:46 bhodi.url
-rw-r--r-- 1 pi pi  165675008 Mar  5 10:52 geexbox-3.1-i386.iso
-rw-r--r-- 1 pi pi          0 Mar  8 13:46 geexbox-3.1-i386.url
drwxr-xr-x 2 pi pi      32768 Feb 26 09:26 iso
-rw-r--r-- 1 pi pi 1091567616 Mar  5 11:05 linux-lite-3.8-32bit.iso
-rw-r--r-- 1 pi pi          0 Mar  8 13:46 linux-lite-3.8-32bit.url
----------------------
This is what I get with: LINUX_LITE_SRC_ISO=/media/pi/3032-6134         
handle_iso(win-pe-x86)
handle_iso(linux_lite)
handle_kernel(arch-netboot-x64)
handle_pxe()
-----------------------------
This is what I get with: LINUX_LITE_SRC_ISO=/media/pi/3032-6134/
handle_iso(linux_lite)
    download iso image
/media/pi/3032-6134/: Scheme missing.
    create nfs folder

beta-tester, 2019-03-10:
first, that `handle_iso $LINUX_LITE $LINUX_LITE_URL  $LINUX_LITE_SRC_ISO $LINUX_LITE_FILE_ISO;` is not making sense, unless you did modified the `handle_iso() { ... }` part in the `...pass2.sh` file to that specific purpose.
maybe those extra `$LINUX_LITE_SRC_ISO $LINUX_LITE_FILE_ISO` parameters you added were added to the `/etc/fstab` file, what may cause that error message.
the fstab entry should look like:
`/srv/iso/linux-lite-x86.iso  /srv/nfs/linux-lite-x86  auto  ro,nofail,auto,loop  0  10`

if the `/etc/fstab` entry for the iso is already wrong, then everything behind is not working...
no NFS for that iso, nor TFTP, nor HTTP and no menu entry will be generated, because it can't see the content of the iso and the expected files.

the `.iso` and `.url` file **must** be at `$SRC_ISO` or at `$DST_ISO` location with the correct nameing according the given `$LINUX_LITE` variable. otherwise the script screws up.

second, did you tried just to use the "normal" way with downloading the iso from the url?
(i know that mirror is far away from your location, but this is the url that worked for me to download with `wget`)
```
LINUX_LITE=linux_lite
LINUX_LITE_URL=http://rwthaachen.dl.osdn.jp/storage/g/l/li/linuxlite/3.8/linux-lite-3.8-32bit.iso
```

```
#========== BEGIN ==========
if [ -f "$FILE_MENU" ] \
&& [ -f "$DST_NFS_ETH0/$LINUX_LITE/casper/vmlinuz" ]; then
    echo  -e "\e[36m    add $LINUX_LITE\e[0m";
    sudo sh -c "cat << EOF  >> $FILE_MENU
    ########################################
    LABEL $LINUX_LITE
        MENU LABEL Linux Lite (32bit)
        KERNEL $FILE_BASE$NFS_ETH0/$LINUX_LITE/casper/vmlinuz
        INITRD $FILE_BASE$NFS_ETH0/$LINUX_LITE/casper/initrd.gz
        APPEND nfsroot=$IP_ETH0:$DST_NFS_ETH0/$LINUX_LITE ro netboot=nfs file=/cdrom/preseed/custom.seed boot=casper --
        TEXT HELP
            Boot to Linux Lite x86 Live
            User: linux
        ENDTEXT
EOF";
fi
#=========== END ===========
```

```
handle_iso $LINUX_LITE  $LINUX_LITE_URL;
```