beta-tester/RPi-PXE-Server #14: Raspberry pi 1 second part installation error

PaulRobert1, 2019-03-15:
Hello, when I try to install pass2.sh  i get ip adress not found error, i have lan setup and tested on it. Please help
Kind Regards.

beta-tester, 2019-03-15:
hello, do you have a LAN cable connected to the RPi 1 and does your RPi 1 got an IP address from that network?

beta-tester, 2019-03-15:
can you please download/clone tha project again...
at least the file p2-include-var.sh ...
i added some code to gather the name of the networkadapter for RPi1, i didn't tested my scripts on an RPi1 before.
hope it will work for you now.

PaulRobert1, 2019-03-15:
Thank you for the fast response, i am going to test it now, yes i have cable connected to lan, ifconfig shows valid ip, and can navigate internet

PaulRobert1, 2019-03-15:
tested it, now when I run part 2 I just get a > and i can type after it, nothing else happens

beta-tester, 2019-03-15:
do you mean part2 script was running without errors,
but and after the script exited, your RPi installation is broken?

PaulRobert1, 2019-03-15:
No, when i ran part 2 the 1st time i downloaded the project, it said no ip
found, and incomplete command error, the second time i dowloaded it when i
run part 2 , i get a new line in the terminal with >
No installation of part 2 whatsoever.

PaulRobert1, 2019-03-15:
https://drive.google.com/file/d/1ac5ow9Al4Ye5zS1UG_ELjUgMc9AD-Zst/view?usp=drivesdk

PaulRobert1, 2019-03-15:
A screenshot of the error is in the attachment above

PaulRobert1, 2019-03-15:
I get the same error when trying to install with bash

beta-tester, 2019-03-15:
thats strange...

i just tested the script on a RPi1B rev2 (512MB) and RPi1B+ and it executed successfully.
i used the actual Raspbian version [Raspbian Stretch with desktop](https://www.raspberrypi.org/downloads/raspbian/)
the Desktop version is maybe too heavy for an RPi1, it run extremly slow to finish.
what version are you using?

the script should not do any interaction with the user.
and at the moment, i do not see, where that `>` could come from.

EDIT: eh... saw your screenshow...
i see you are running the script as root. please run the script without root (su).
`bash install-pxe-server_pass2.sh`

PaulRobert1, 2019-03-15:
I tried with root, without root, with bash, with ./
I have raspbian stretch the basic, not the full one, downloaded today

beta-tester, 2019-03-15:
this is the same version i used.
what version of RPi1 do you have RPi1B rev.1 with 256MB (short GPIO header), a RPi1B rev2 with 512MB (short GPIO header) or a RPi1B+ with 512MB and long GPIO header?

if nothing will help, try to edit the p2-include-var.sh file and put the correct interface nam at line 33:
```
INTERFACE_ETH0=eth0
```
in case `eth0` is not your network interface name, please use the right name.

PaulRobert1, 2019-03-15:
Rpi 1 short gpio header, model b 512mb

PaulRobert1, 2019-03-15:
Thank you for your help, i will check and see.

beta-tester, 2019-03-15:
i will test the script again... this will take a wile.
in the meantime please modify the p2-include-var.sh file and add `eth0` as shown above.

PaulRobert1, 2019-03-15:
Ok

beta-tester, 2019-03-15:
i am out of ideas.

i added a "fallback" to the script to use `eth0` in case it could not gather name of the interface from the system.
but normally it should have worked already with the first version of the fix.

if it is still not working for you.

- make sure your `p2-include-var.sh` is the newest version (should have the `# fallback` included).
- make sure the `p2-include-var.sh` is in the same folder as the `install-pxe-server_pass2.sh` file.
- make sure you don't run the script as root (no su, no sudo).
- what is `find /sys/devices/ -type d | grep "1:1.0/net/"` giving you?

PaulRobert1, 2019-03-16:
your fix worked, thank you very much! i am now downloading the os list selected, if say i want to add a windows 10 iso what is the procedure? I put the iso in the iso folder, add the name of the iso in handle.sh and re run part 2?

beta-tester, 2019-03-16:
thank you for your report and good to hear, that it works.

for the windows 10 iso,
do you mean a windows 10 PE iso that is around 500MByte big,
or do you mean a whole windows 10 installation iso that is about 3GByte big?

if you have a Windows PE iso it is easy to integrate to the PXE boot process. you can add it as memdisk - it is not the most elegant way and eats twice the RAM as necessary, but it is easy.
from within the Win PE, you can mount a (samba) network share to a windows 10 installation media and install windows via that share.

to PXE boot the full Windows 10 installation iso will be complicated, because of its size. it can't be loaded as a memdisk anmore.
and because of you have to tell windows where to find the installation files on the network.

see also:
[Download and install the Windows ADK](https://docs.microsoft.com/de-de/windows-hardware/get-started/adk-install) to create you own Windows PE iso.
[Windows PE (WinPE)](https://docs.microsoft.com/en-us/windows-hardware/manufacture/desktop/winpe-intro)
[Configure a PXE server to load Windows PE](https://docs.microsoft.com/en-us/windows/deployment/configure-a-pxe-server-to-load-windows-pe)

a template for Windows PE (ISO) is already in the script. you have to provide a windows PE iso that must have the name `win-pe-x86.iso` and an empty `win-pe-x86.url` file. copy both to the destination folder, where all the ISOs were stored and re-run the pass2 script. it should see the win-pe-x86.iso and integrate it to the PXE-menu.

note, that the architecture (x86 or amd64) of Windows PE and Windows 10 installation must be the same, when you try to install Windows from within Windows PE.
if you only want to get access to an already installed Windows system, you can use either windows  x86 or amd64 architecture. then it doesn't matter. i use the x86 version, because there are more applications available that runs on x86 and are not aveilable for amd64.

maybe you open a new issue with a request to windows PE.
then i will try to explain it there for the public or will add a wiki.

closed 2019-03-16 by beta-tester