beta-tester/RPi-PXE-Server #15: PXE Server on boot and adding whole windows 10 installation media

PaulRobert1, 2019-03-16:
Hello,

I have sucessfully installed part 1 and 2, do I need to configure annything else? and I can t find annything in the processes list with ps -A. Does the server now start at boot? I tried hooking it up to my Router and launch PXE boot on my laptop, but I get no file recieved. And how do I configure the server?

Regarding Windows 10, it would be amazing to to keep all your installation media say on a 64 gig drive in your RPI and install them trough your Network. Say my old windows 7 Pc don t boot, just boot the installation kit trough LAN :)

Thank you for the fast and helpful responses:)

Kind Regards.

beta-tester, 2019-03-16:
to the first part of your question.
yes, the PXE-/DHCP-/DNS-/TFTP-/HTTP-/NFS-/SMB- server/-services are started just after every boot automatically.
the first four services (PXE-/DHCP-/DNS-/TFTP-) were done by `dnsmasqe`, the NFS is done by `nfs-kernel-server`, SMB- by `samba` and HTTP by `lighttpd`.

each iso that is included in the script were tested to run. if an iso isn't in the PXE-menu visible, check the `p2-include-handle.sh` file, that system you want to see in the pxe menu has a `handle ...` line. `_unhandle ...` will remove an iso if it was previously handled. a out commented line will ignore an iso (if it was handled befor, it stays in the pxe-menu).
normally nothing else have to be configured after the pass2 script were ran /re-runned.

after the pass2 script was run, the PXE-Server can act as a stand alone server in an isolated network (you don't have a connection to external internet then)
or in co-existence to an existing router as gateway to internet.
everything connected to the RPi's internet adapter will see the RPi as PXE-server through that network.

note: only BIOS pxe is tested. UEFI IPv4 and UEFI IPv6 i could not test, because i dont have those hadtware.

to the second part of your question.
please give me time to figure out if i can find a nice way to describe the procedure of including windows 10 installationmedia.

the easiest procedure i suggest is to make a 32bit and a 64bit version of a Windows PE iso that the client computer can pxe boot into.
the script has a template for win-pe-x86 already included. search for `WIN_PE_X86` on the scripts `p2-include-url.sh`, `p2-include-menu.sh` and `p2-include-handle.sh`.
if you have a winPE iso, rename it to `win-pe-x86.iso`, create an empty `win-pe-x86.url` file and copy both to the `/srv/iso` destination folder. (for a 64bit version copy/paste all the `WIN_PE_X86` stuff and adapt it to `WIN_PE_X64` for example.)
put all the different windows 10 installation files accessable via SMB on the PXE server.
e.g.: copy all installation files to a folder `/srv/windows10_installation_files` on the server.
once the client has pxe booted the windows PE system, it can activate its network.
and mount the share where your the windows 10 installation files are stored.
something like:
```
wpeutil InitializeNetwork
net use N: \\pxe-server\srv guest /USER:guest
N:
cd windows10_install_files
setup.exe
```

for more information:
[Download and install the Windows ADK](https://docs.microsoft.com/de-de/windows-hardware/get-started/adk-install) to create you own Windows PE iso. you nead at least the "Windows PE add-on for the ADK" to biuld your WinPE iso.
[Windows PE (WinPE)](https://docs.microsoft.com/en-us/windows-hardware/manufacture/desktop/winpe-intro)

i will put it on my todo list for a better description.

PaulRobert1, 2019-03-16:
unfortunetly i still coudnt get another computer to boot over PXE, how can I check if everything is setup ok, example after i installed part 1 and 2 i noticed another network interface showing up in wired network settings, but in terminal with ifconfig nothing shows up, my pi is currently hooked up to my network and i am accesing it trough VNC.

beta-tester, 2019-03-16:
what are the names of the network interfaces and which of the two are shown up in `ifconfig`?

to see if the pxe server is working, open a terminal window on the server and type in:
```
tail -F /var/log/syslog
```
so you can watch the logfile, while the PC is booting up.
you should see some dnsmasq logs:
```
dnsmasq-dhcp[.]:  available DHCP range: 192.168.1.200 -- 192.168.1.250
dnsmasq-dhcp[.]:  client provides name: XXXXXXXX
dnsmasq-dhcp[.]:  available DHCP range: 192.168.1.200 -- 192.168.1.250
dnsmasq-dhcp[.]:  vendor class: PXEClient:Arch:00000:UNDI:002001
dnsmasq-dhcp[.]:  DHCPDISCOVER(eth0) XX:XX:XX:XX:XX:XX
dnsmasq-dhcp[.]:  tags: x86_BIOS, ARCH_0, eth0
dnsmasq-dhcp[.]:  DHCPOFFER(eth0) 192.168.1.217 XX:XX:XX:XX:XX:XX
dnsmasq-dhcp[.]:  requested options: 1:netmask, 2:time-offset, 3:router, 4, 5,
dnsmasq-dhcp[.]:  requested options: 6:dns-server, 11, 12:hostname, 13:boot-file-size,
dnsmasq-dhcp[.]:  requested options: 15:domain-name, 16:swap-server, 17:root-path,
dnsmasq-dhcp[.]:  requested options: 18:extension-path, 22:max-datagram-reassembly,
dnsmasq-dhcp[.]:  requested options: 23:default-ttl, 28:broadcast, 40:nis-domain,
dnsmasq-dhcp[.]:  requested options: 41:nis-server, 42:ntp-server, 43:vendor-encap,
dnsmasq-dhcp[.]:  requested options: 50:requested-address, 51:lease-time, 54:server-identifier,
dnsmasq-dhcp[.]:  requested options: 58:T1, 59:T2, 60:vendor-class, 66:tftp-server,
dnsmasq-dhcp[.]:  requested options: 67:bootfile-name, 128, 129, 130, 131, 132,
dnsmasq-dhcp[.]:  requested options: 133, 134, 135
dnsmasq-dhcp[.]:  next server: 192.168.1.100
dnsmasq-dhcp[.]:  broadcast response
dnsmasq-dhcp[.]:  sent size:  1 option: 53 message-type  2
dnsmasq-dhcp[.]:  sent size:  4 option: 54 server-identifier  192.168.1.100
dnsmasq-dhcp[.]:  sent size:  4 option: 51 lease-time  1d
dnsmasq-dhcp[.]:  sent size: 22 option: 67 bootfile-name  menu-bios/lpxelinux.0
dnsmasq-dhcp[.]:  sent size:  4 option: 58 T1  12h
dnsmasq-dhcp[.]:  sent size:  4 option: 59 T2  21h
dnsmasq-dhcp[.]:  sent size:  4 option:  1 netmask  255.255.255.0
dnsmasq-dhcp[.]:  sent size:  4 option: 28 broadcast  192.168.1.255
dnsmasq-dhcp[.]:  sent size:  4 option:  3 router  192.168.1.100
dnsmasq-dhcp[.]:  sent size:  4 option:  6 dns-server  192.168.1.100
dnsmasq-dhcp[.]:  sent size:  4 option: 42 ntp-server  192.168.1.100
dnsmasq-dhcp[.]:  sent size:  8 option: 66 tftp-server  0.0.0.0
dnsmasq-dhcp[.]:  sent size:  9 option: 60 vendor-class  XX:XX:XX:XX:XX:XX:XX:XX:XX
dnsmasq-dhcp[.]:  sent size: 17 option: 97 client-machine-id  XX:XX:XX:XX:XX:XX:XX:XX:XX
dnsmasq-dhcp[.]:  sent size: 50 option: 43 vendor-encap  XX:XX:XX:XX:XX:XX:XX:XX:XX
dnsmasq-dhcp[.]:  available DHCP range: 192.168.1.200 -- 192.168.1.250
dnsmasq-dhcp[.]:  vendor class: PXEClient:Arch:00000:UNDI:002001
dnsmasq-dhcp[.]:  DHCPDISCOVER(eth0) XX:XX:XX:XX:XX:XX
dnsmasq-dhcp[.]:  tags: x86_BIOS, ARCH_0, eth0
dnsmasq-dhcp[.]:  DHCPOFFER(eth0) 192.168.1.217 XX:XX:XX:XX:XX:XX
dnsmasq-dhcp[.]:  requested options: 1:netmask, 2:time-offset, 3:router, 4, 5,
dnsmasq-dhcp[.]:  requested options: 6:dns-server, 11, 12:hostname, 13:boot-file-size,
dnsmasq-dhcp[.]:  requested options: 15:domain-name, 16:swap-server, 17:root-path,
dnsmasq-dhcp[.]:  requested options: 18:extension-path, 22:max-datagram-reassembly,
dnsmasq-dhcp[.]:  requested options: 23:default-ttl, 28:broadcast, 40:nis-domain,
dnsmasq-dhcp[.]:  requested options: 41:nis-server, 42:ntp-server, 43:vendor-encap,
dnsmasq-dhcp[.]:  requested options: 50:requested-address, 51:lease-time, 54:server-identifier,
dnsmasq-dhcp[.]:  requested options: 58:T1, 59:T2, 60:vendor-class, 66:tftp-server,
dnsmasq-dhcp[.]:  requested options: 67:bootfile-name, 128, 129, 130, 131, 132,
dnsmasq-dhcp[.]:  requested options: 133, 134, 135
dnsmasq-dhcp[.]:  next server: 192.168.1.100
dnsmasq-dhcp[.]:  broadcast response
dnsmasq-dhcp[.]:  sent size:  1 option: 53 message-type  2
dnsmasq-dhcp[.]:  sent size:  4 option: 54 server-identifier  192.168.1.100
dnsmasq-dhcp[.]:  sent size:  4 option: 51 lease-time  1d
dnsmasq-dhcp[.]:  sent size: 22 option: 67 bootfile-name  menu-bios/lpxelinux.0
dnsmasq-dhcp[.]:  sent size:  4 option: 58 T1  12h
dnsmasq-dhcp[.]:  sent size:  4 option: 59 T2  21h
dnsmasq-dhcp[.]:  sent size:  4 option:  1 netmask  255.255.255.0
dnsmasq-dhcp[.]:  sent size:  4 option: 28 broadcast  192.168.1.255
dnsmasq-dhcp[.]:  sent size:  4 option:  3 router  192.168.1.100
dnsmasq-dhcp[.]:  sent size:  4 option:  6 dns-server  192.168.1.100
[... 63 lines cut ...]
dnsmasq-tftp[.]: sent /srv/tftp/menu-bios/pxelinux.cfg/default to 192.168.1.217
dnsmasq-tftp[.]: sent /srv/tftp/menu-bios/vesamenu.c32 to 192.168.1.217
dnsmasq-tftp[.]: sent /srv/tftp/menu-bios/libcom32.c32 to 192.168.1.217
dnsmasq-tftp[.]: sent /srv/tftp/menu-bios/libutil.c32 to 192.168.1.217
dnsmasq-tftp[.]: sent /srv/tftp/menu-bios/pxelinux.cfg/default to 192.168.1.217
```
at this point the client pc should show the pxe menu.

the most important lines are:
```
dnsmasq-dhcp[.]:  available DHCP range: 192.168.1.200 -- 192.168.1.250
...
dnsmasq-dhcp[.]:  DHCPOFFER(eth0) 192.168.1.217 XX:XX:XX:XX:XX:XX
...
dnsmasq-dhcp[.]:  sent size: 22 option: 67 bootfile-name  menu-bios/lpxelinux.0
...
dnsmasq-dhcp[.]:  PXE(eth0) 192.168.1.217 XX:XX:XX:XX:XX:XX menu-bios/lpxelinux.0

dnsmasq-tftp[.]: sent /srv/tftp/menu-bios/lpxelinux.0 to 192.168.1.217
...
dnsmasq-tftp[.]: sent /srv/tftp/menu-bios/pxelinux.cfg/default to 192.168.1.217

```
where 192.168.1.217 the IP of the client pc is that the PXE server was giving to him.
and XX:XX:XX:XX:XX:XX the mac address of the client pc.

beta-tester, 2019-03-16:
to check if the services are running:
```
sudo systemctl status dnsmasq
sudo systemctl status nfs-kernel-server
sudo systemctl status lighttpd
sudo systemctl status smbd
```
all should show something like that the service is enabled, active and running.

PaulRobert1, 2019-03-17:
https://drive.google.com/open?id=18dL3lyw6TKFiLxYnrXp_H8QDsGw1HeDE

unfortunetly, after issuing the command tail -F /var/log/syslog I get nothing similar to what you sent me

beta-tester, 2019-03-17:
> 
> 
> unfortunetly, after issuing the command tail -F /var/log/syslog I get nothing similar to what you sent me

did you booted up the client PC, after issuing that commandline?
did you also tried the  commands of the other post as well?

> 
> 
> example after i installed part 1 and 2 i noticed another network interface showing up in wired network settings, but in terminal with ifconfig that new connection dosnt show up

and what are the names of the two network interfaces?

PaulRobert1, 2019-03-17:
This morning i did this: reflashed my pi fresh installed the 2 parts,  with no results, one network interface is eth0 the other disappeared, it was called et052bc....something, found that odd, after i launched the syalog command, tried to boot up client pc, with no results, would you be willing to teamviewer into my pc? Lol

beta-tester, 2019-03-17:
you have teamviewer installed on the RPi1?
sure, i can do, if you want.
but it means you have to show your teamviewer ID and the password to the public.
(or does github has a PrivateMessage function?)

PaulRobert1, 2019-03-17:
I have teamviewer on my pc wich is vnc d to my teamviewer i suggest you
give me your email so i can send my id and pass  there :)

PaulRobert1, 2019-03-17:
or here s mine, robertlorinczi@gmail.com, write me a reply:)

beta-tester, 2019-03-17:
i have a 10 minute email addres: q1410816@nwytg.net

PaulRobert1, 2019-03-17:
sent

PaulRobert1, 2019-03-17:
Hab aus versehen, strohm schalter gedruckt als ich den rpi entfernt
habe:))))

beta-tester, 2019-03-17:
connection lost

PaulRobert1, 2019-03-17:
Accidentally switched off main power lmfao

PaulRobert1, 2019-03-17:
Try now

PaulRobert1, 2019-03-17:
The teamviewer password changed, is that email still valid?

beta-tester, 2019-03-17:
:)
q1413931@nwytg.net meine neue email  für die nachsten 10 minuten.

PaulRobert1, 2019-03-17:
sent

PaulRobert1, 2019-03-17:
Thank you very much for the Remote Support! I tried trough the router, and it boots!!!!

Have a great day.

beta-tester, 2019-03-17:
you are welcome.
i deleted all posts with all the eMails.

conclusion is, the router is eating all the dhcp traffic, when the PXE-server and the client PC are connected to the router.
maybe put a network switch to the router and connect the RPi and the client PC to the switch.

PaulRobert1, 2019-03-17:
this is exactly what I did, earlier the RPI was hooked up to the Router, now it s hooked up to the switch
Thank you for your awsome support!

beta-tester, 2019-03-17:
i searched for the manual of your router...
there are options they could control the behaviour:

> 3.4.3  DHCP Server Configuration
> 
> - Enable DHCP L2 Relay
> - Enable secondary DHCP server

but i am not sure.

closed 2019-03-17 by beta-tester

PaulRobert1, 2019-03-17:
Most likely the second dhcp server, but it works now, hey what do you say
to this idea: creating a grub bootloader, bootable image, with all your
apps and os( win7,8,xp, etc), and installing it on the rpi, would that
work?:)

beta-tester, 2019-03-17:
with grub bootloader i don't have any experience.

PaulRobert1, 2019-03-17:
Gonna try see what happens:)

beta-tester, 2019-03-17:
if it works, let me know. :)

PaulRobert1, 2019-03-17:
Oh, rest assured, you will be the 1st one to know :D

AlbertoHugonin, 2019-09-22:
Is there a way to boot directly into windows 10 installation instead of using windowspe?
This is my script folder, i already set up the url and the handle scripts.
[Rpi-PXE.zip](https://github.com/beta-tester/RPi-PXE-Server/files/3640241/Rpi-PXE.zip)
Hope you can help

beta-tester, 2019-09-23:
@AlbertoHugonin, it is _maybe_ possible (not tested), but for sure not without modifications.
1. you can not doing it by loading the ISO as ramdisk, because of the enormous huge size of the ISO and because of windows can not see the files inside the ramdisk.
2. you have to use a "_pxeboot.n12_" boot file i am not allowed provide. you have to temporarily install Microsoft ADK + WinPE to get the needed file. see [Configure a PXE server to load Windows PE](https://docs.microsoft.com/en-us/windows/deployment/configure-a-pxe-server-to-load-windows-pe)
the file "pxeboot.n12" is in fact inside the boot.wim
(<mount_point>\Windows\Boot\PXE\pxeboot.n12)
3. you have to copy some other files to the TFTP server
```
.
├── Boot
│   ├── BCD
│   ├── Fonts
│   │   └── wgl4_boot.ttf
│   └── boot.sdi
├── bootmgr.exe
├── sources -> /srv/nfs/Win10_1903_V1_Italian_x64/sources/
```
4. you have to modify the BCD file:
```
bcdedit /store %1 /create {99999999-9999-9999-9999-999999999999} /application osloader
bcdedit /store %1 /set {99999999-9999-9999-9999-999999999999} description     "Windows PE (64-bit)"
bcdedit /store %1 /set {99999999-9999-9999-9999-999999999999} device          ramdisk=[boot]\win-pe-x64\sources\boot.wim,{ramdiskoptions}
bcdedit /store %1 /set {99999999-9999-9999-9999-999999999999} osdevice        ramdisk=[boot]\win-pe-x64\sources\boot.wim,{ramdiskoptions}
bcdedit /store %1 /set {99999999-9999-9999-9999-999999999999} systemroot      \Windows
bcdedit /store %1 /set {99999999-9999-9999-9999-999999999999} path            \Windows\System32\Boot\winload.exe
bcdedit /store %1 /set {99999999-9999-9999-9999-999999999999} locale          en-US
bcdedit /store %1 /set {99999999-9999-9999-9999-999999999999} inherit         {bootloadersettings} /addlast
bcdedit /store %1 /set {99999999-9999-9999-9999-999999999999} isolatedcontext Yes
bcdedit /store %1 /set {99999999-9999-9999-9999-999999999999} bootmenupolicy  Legacy
bcdedit /store %1 /set {99999999-9999-9999-9999-999999999999} detecthal       Yes
bcdedit /store %1 /set {99999999-9999-9999-9999-999999999999} winpe           Yes
bcdedit /store %1 /set {99999999-9999-9999-9999-999999999999} ems             No
bcdedit /store %1 /set {bootmgr} displayorder {99999999-9999-9999-9999-999999999999} /addlast
```

note: the steps above roughly describes only **BIOS** like PXE.
for UEFI like PXE you have also to provide the EFI files on TFTP and BCD configuration for EFI.

and i have no idea, how to modify the configuration of the windows setup to tell its setup, where to find the neccessary files for setup on the network.

so maybe it is easier to prepare a WinPE iso that will then start the setup of the mountes windows installation iso.
you can create a script in the WinPE iso, that will autostart.
(initialize network via `wpeutils initiaize network`, mount the installation ISO via `net use I: \\<ip_of_your_server>\nfs\Win10_1903_V1_Italian_x64 user=guest password=guest`, execute `I:\setup.exe` )

BTW: when you set `WIN=Win10_1903_V1_Italian_x64`, then you have to add also a corresponding counterpart in p2-include-menu.sh
```
#========== BEGIN ==========
if [ -f "$FILE_MENU" ] \
&& [ -f "$DST_TFTP_ETH0/$1/pxeboot.n12" ]; then
    echo  -e "\e[36m    add $WIN (PXE)\e[0m";
    sudo sh -c "cat << EOF  >> $FILE_MENU
    ########################################
    LABEL $WIN-pxe
        MENU LABEL Windows PE x64 (PXE)
        PXE pxeboot.n12
        TEXT HELP
            Boot to Windows PE 64bit
        ENDTEXT
EOF";
fi
#=========== END ===========

#========== BEGIN ==========
if [ -f "$FILE_MENU" ] \
&& [ -f "$DST_ISO/$WIN.iso" ]; then
    echo  -e "\e[36m    add $WIN (ISO)\e[0m";
    sudo sh -c "cat << EOF  >> $FILE_MENU
    ########################################
    LABEL $WIN-iso
        MENU LABEL Windows PE x64 (ISO)
        KERNEL memdisk
        APPEND iso
        INITRD $FILE_BASE$ISO/$WIN.iso
        TEXT HELP
            Boot to Windows PE 64bit ISO ~400MB
        ENDTEXT
EOF";
fi
#=========== END ===========
```

AlbertoHugonin, 2019-09-23:
@beta-tester thanks you very much for the indications you gave me, but for the sake of simplicity (because i nedeed to install windows as soon as possible) i used a program called AIO Boot that did the job very quickly. Acording to the site: "AIO Boot will modify the boot.wim file to interfere with its working steps. Or we will direct the path of the install.wim file". So maybe you can find something usefull. I will use your program for other distributions for now.
Also thanks for the hard work

beta-tester, 2019-09-23:
i guess it will not help you, but i just added wimboot.
see https://github.com/beta-tester/RPi-PXE-Server/commit/6b89500b455cbb12876f5dd37687c69ee42e52cf
i realized that using wimboot is by far the quickest boot option for WinPE.
(no need to modify BCD or using pxeboot.n12, just uses the mounted ISO)