beta-tester/RPi-PXE-Server #16: Future request: Docker Container

squadramunter, 2019-03-23:
Hi, First I would thank you for this amazing project you've done.
It helps me with installing my distro's a lot easier.

Can you make a docker container from this? I don't want to mess things up if it goes wrong. And is it safe to be used when other services are running that are using port 53? Like dnsmasq or bind9?

Thanks and hope you can help me with this question.

beta-tester, 2019-03-25:
hello,
at the moment i have absolutely no knowledge about how to setup and use docker.
if you have a link, where i can find an easy step by step manual, i will see, what i can do for you.

to setup a PXE server with using dnsmasq, you have to enable at lease TFTP, give the client an P address and tell him that you are a tftp-server and that there is a bootloader file and provide the bootloader file via tftp.

```
# /etc/dnsmasq.d/my_config_file

# enable TFTP
enable-tftp

# where is the root of TFTP folder
tftp-root=/srv/tftp/

# tell the client, that you are a TFTP server, when the client is asking for. (0.0.0.0 or the IP of your server)
dhcp-option=option:tftp-server, 0.0.0.0

# give out an IP address, when the client is asking for (DHCP request)
# do not give IPs that are in pool of DSL routers DHCP
dhcp-range=192.168.1.200, 192.168.1.250, 24h

# tell what/where the bootloader file is, when the client is asking for
dhcp-boot=lpxelinux.0

# set the pxe menu bootloader
pxe-service=x86PC, "PXE Boot Menu (BIOS 00:00)", lpxelinux
```

at least the bootloader file (in this case the lpxelinux as pxe-menu bootloader) has to be in the /srv/tftp folder.

see documentation:
http://www.thekelleys.org.uk/dnsmasq/docs/dnsmasq-man.html
https://wiki.syslinux.org/wiki/index.php?title=PXELINUX-Multi-Arch
https://wiki.syslinux.org/wiki/index.php?title=SYSLINUX#APPEND

because tftp is extremely slow, it it recommend to use an other possibility to provide files after the bootloader was loaded by the client. http-server, nfs-server or others, depending on, what the bootloader is able to handle. lpxelinux can handle nfs and http very well. but can not domainnames. you have to provide always IP adresses in APPEND and INITRD.

and then it depends what the loaded kernel is able to handle http, nfs, smb, ATA over Ethernet, ...

you can use apache as http server instead of lighttps i used.

for more, i have to study documents... specially the docker thing.

squadramunter, 2019-04-22:
Creating a docker container isn't that hard. ofc your need to deep into it and take some time to learn how it works. But it is as simple as writing bash code.

https://docs.docker.com/get-started/

squadramunter, 2019-04-23:
I tried to make your script working on Debian 9 Stretch X64 and I can say that it is working perfect! The only thing that isn't working is UEFI boot but I don't know if this is a problem in your code or the way I installed it on a 64 bit system.

beta-tester, 2019-07-15:
i close that issue, because i don't make a docker container.

closed 2019-07-15 by beta-tester