beta-tester/RPi-PXE-Server #19: UEFI Boot is not working anymore

squadramunter, 2019-04-23:
I have tried to boot with UEFI mode in BIOS settings but it did not work. I use Reaktek controller to boot from it. My laptop is Asus K501LX. Booting from Legacy (CSM) does work but UEFI doesn't.

_Originally posted by @squadramunter in https://github.com/beta-tester/RPi-PXE-Server/issues/6#issuecomment-485731638_

squadramunter, 2019-04-23:
Maybe that this url can help to investigate the problem?
https://wiki.ubuntu.com/UEFI/SecureBoot/PXE-IPv6?action=show&redirect=UEFI%2FSecureBoot-PXE-IPv6

beta-tester, 2019-04-23:
hello @squadramunter, than you for the urls...

i am not sure if UEFI IPv4 or UEFI IPv6 was ever working at all in my PXE-Server project, because i never could test it by myself - i don't have hardware that supports UEFI IPv4/UEFI IPv6.
and i am pretty sure when SecureBoot is enabled it will definitely not work. none of the SYSLINUX bootloaders are signed by microsoft UEFI certificate properly.

beta-tester, 2019-04-23:
BTW: i only have a Windows 10 netbook that isn't able to PXE boot UEFI IPv4 nor UEFI IPv6.
my netbook has problems to UEFI boot Linux distributions from USB-memory-stick/USB-DVD-drive, because is has SecureBoot enabled and does not allow to disable it and it has only UEFI 32 with a 32/64bit CPU.
but i know that only Fedora 29 has proper Microsoft UEFI CA signed bootloader on their ISO image.
for UEFI 64 and UEFI 32. ubuntu isn't intended to add a UEFI 32 to their 64bit ISO.
debian still does not have added a proper Microsoft UEFI CA signed UEFI 32 bootloader on their ISOs.

at the moment Fedora 29 Workstation (64bit) is the only Linux distribution i tried, that is able to boot my netbook without disabling SecureBoot.

squadramunter, 2019-05-02:
Never enable SecureBoot on a Linux installation because it can breaks loading Kernel Modules. Like you Wireless card or even your Nvidia graphics card. Only enable secure boot if there is a reason for it. You can setup full disk encryption with LUKS1. If you have SecureBoot enabled and you try to upgrade your kernel the modules fails to load properly.

beta-tester, 2019-05-02:
on that netbook i have no choice to disable SecureBoot.
Windows 10 is installed there with full encryption and i have no intention to install Linux to that netbook.
i only tried to use it to test UEFI boot, but it has no UEFI IPv4/6 option... so it is useless to use it for test purpose.

beta-tester, 2019-07-15:
with debian-buster / raspbian-buster there are newer syslinux packets included.
maybe this will work...
but i still can't test UEFI boot.
maybe you have more luck...

beta-tester, 2019-07-15:
PS.: Debian 10 Buster Live ISO has now UEFI bootloader included that ist properly signed. so disabling SecureBoot shouldn't be neccessary when using that ISO...

HarvsG, 2019-07-18:
@squadramunter these issues may help you #6 #7

hsuanpai, 2019-07-31:
I can help you to test the UEFI boot w/ Server level platform.
I tried and use tshark to monitor and got PXE-E16 of the error message.
Please advise.

beta-tester, 2019-07-31:
hi, thank you @hsuanpai  for trying...
i searched for **PXE-E16** and it seems that the computer isn't getting the boot file name presented by the pxe-server.

can you take a look to the **/var/log/syslog** and search for **dnsmasq-dhcp** logs.
specially "**bootfile name:** ..." entries.
did the pxe-server offered an boot file name?
if so, can you see a file transfer via TFTP?

beta-tester, 2020-11-30:
i could successfully PXE UEFI boot from an ASUS machine while UEFI + SecureBoot is enabled.
only **UEFI: IPv4** is working there, IPv6 isn't.

but it completely fails to do the same from a DELL machine. no idea, what is missing there to get it work.

i tested the current live distros Fedora x64, Ubuntu x64, Debian x64, Mint x64 with success.

ThuGie, 2022-02-09:
Hi,

Is there work being done on UEFI booting ?
As legacy it boots fine, but uefi it looks like the server doesnt exists at all.
Or is there a setting wrong ?
That linux doesnt support secure boot is fine by me, my laptop has it disabled, and plan to try adding windows to it later so.

beta-tester, 2022-02-10:
i can only tell that it will work on one of my hardware...

it is higly depending on the board you are booting from and the OS you want to boot into.
if you do not see the boot menu then the board has some specific requirements i don't know.
if you see the boot menu but the OS you try to boot into then maybe there is no SecureBoot support in that OS or has its own key that is not stored in your board yet.

the busines DELL laptop i never got UEFI booting.
my old private PC with ASUS board can UEFI boot, but i have to hit a key to get the boot menu to see. then i have to select the boot option:
**UEFI Network IPv4**
_UEFI Network IPv6_ is not working.

when **SecureBoot** is enabled i only be able to boot into:
**Debian**, **Fedora**, **OpenSuse**, **Ubuntu** _(maybe i forgot some OS)_
the other OS'es start booting but get stuck later on in boot process.

in case you did not used my script on a fresh RasPi OS installation, please try to do it from a fresh installation.

when you tell that is looks like there is no PXE serer present from the piont of view of your PC when UEFI is enabled, do you see any activity on the `/var/log/syslog` file on the PXE server?
e.g. a message for trying to give the PC an ip address after seeing a request for an IP of your PC

to see live log messages:
`tail -F /var/log/syslog`