beta-tester/RPi-PXE-Server #21: How to create Clover Bootloader on PXE Server?

stonexing, 2019-05-06:
Hello:
Use your script project, I can already boot WinPE, a lots linux system easily.
Now, I want to boot Clover EFI Bootloader from PXE server, So that I can easily install Apple macOS on my Hackintosh PC.
I just test create an ISO file from EFI 
1. [image]
2. Add URL
`WIN_PE_X86=Win10PEx64_WePE2.0_CGI_
WIN_PE_X86_URL=http://192.168.88.100:8000/Win10PEx64_WePE2.0_CGI_.ISO

APPLE_CLOVER=Apple
APPLE_CLOVER_URL=http://192.168.88.100:8000/Apple.iso`

3. Include handle:
`handle_iso  $WIN_PE_X86  $WIN_PE_X86_URL;

handle_iso $APPLE_CLOVER $APPLE_CLOVER_URL;`

After do this, only WinPE is Appeared in PXE Boot menu, and nothing for Clover EFI Bootloader. How can I do that?

beta-tester, 2019-05-06:
you need also to create a PXE menu entry in `p2-include-menu.sh` for CLOVER...
something like:
```
#========== BEGIN ==========
if [ -f "$FILE_MENU" ] \
&& [ -f "$DST_ISO/$APPLE_CLOVER.iso" ]; then
    echo  -e "\e[36m    add $APPLE_CLOVER (ISO)\e[0m";
    sudo sh -c "cat << EOF  >> $FILE_MENU
    ########################################
    LABEL $APPLE_CLOVER-iso
        MENU LABEL APPLE_CLOVER (ISO)
        KERNEL memdisk
        APPEND iso
        INITRD $FILE_BASE$ISO/$APPLE_CLOVER.iso
        TEXT HELP
            Boot to APPLE_CLOVER ISO ~400MB
        ENDTEXT
EOF";
fi
#=========== END ===========

```

but be aware, that the memdisk method has annoying limits.
e.g.
1. the size of the iso is limited.
2. as soon the cpu is switched to protected mode, it looses the view to the iso in ram and the possibility to access to the iso.

i don't know the apple clover iso... where can i download it? maybe i can figure out how it can be PXE booted and add it to the project.

stonexing, 2019-05-07:
Many Thank's for @beta-tester 
Now I Can boot Clover.iso (origin image download for web) from PXE well.
But after I modify Clover.iso (Add some macOS driver such as APFS driver) for my Hackintosh PC.
I got this message:
[image]
[image]

. 

Origin [Clover.iso](https://jxjjxy-my.sharepoint.com/:u:/g/personal/fktgfcjcp_t_odmail_cn/EZZ-tdRkp5xCj38WrH6dVNMB-sDZvUKaB0OLpvCitK9o4Q?e=hjbYNN)
Modified [CloverB15.iso](https://jxjjxy-my.sharepoint.com/:u:/g/personal/fktgfcjcp_t_odmail_cn/Eaoa9KGy2mdLr4UgQ6_11lQBCajizpz1Gslz3Ed9wCuKBw?e=V7yibK)

Both of those 2 iso can boot after burn them to USB flash driver.
But only origin Clover.iso can boot sucess by PXE method.
How can I make CloverB15.iso bootable from PEX?

beta-tester, 2019-05-07:
when i open those ISO fileswith [7-zip](https://www.7-zip.org/) then i can see, that your modified ISO does have a subfolder where all the files are sitting. so i guess it doesn't have the necessary origin structure to get booted well.
and your image does not have a [BOOT] loader part as the origin ISO has.

```
Clover.iso
├── EFI
|   └── BOOT
|       └── BOOTX64.EFI
├── [BOOT]
|   └── Boot-NoEmul.img
```

```
CloverB15.iso
├── Clover
|   ├── EFI
|   |   └── BOOT
|   |       └── BOOTX64.EFI
```
i gues the tools you used to create your ISO is not suitable for your purpose. it does not create a legacy bootable ISO (i think the [BOOT] folder shown in 7Zip is something like El Torito or something similar)...
see [El Torito CD-ROM standard](https://en.wikipedia.org/wiki/El_Torito_(CD-ROM_standard))

beta-tester, 2019-05-07:
maybe you have more luck with additional option 'raw' in the menu 
`APPEND iso raw`

stonexing, 2019-05-08:
Thank you very much for the helping!
I find out how to build Clover project.
And I Add my custom driver file when building project,
Then the Clover.iso I got can be booting from PXE server well.
The project building Clover is here:[A script to build CloverV2 bootloader under macOS X and Ubuntu 16.04 +](https://github.com/Micky1979/Build_Clover)
I modifed and add my file before building project.
So get get the [Clover.iso](https://jxjjxy-my.sharepoint.com/:u:/g/personal/fktgfcjcp_t_odmail_cn/EXV1Ep7ujQNEu9wAzNqyq0gBLCiL6O3Ahb7byzLktrRpbQ?e=jQSHby) with my custom hackintosh drivers.
And it can be boot well. 
even it still not what I want, but at least it has been modified and can be boot form PXE.

closed 2019-05-08 by stonexing