beta-tester/RPi-PXE-Server #23: About Add ISOS

mdiharce, 2019-05-30:
Hello, first of all thank you for such a useful development, it is excellent! I wanted to ask you if there is any tutorial to add isos type medicat, Windows 7 AIO or installers of W10 (Preferably for novices). Excuse me if you already explained it but I'm new with this, and many things happen to me, but I feel like it and force :)
Regards!

beta-tester, 2019-07-15:
i am so sorry... it is not easy to explain the way to go.
and the way i know will result in a very basic WinPE image with only a console window.

at the moment i can only provide the official link to [Get Started - Windows ADK](https://docs.microsoft.com/en-us/windows-hardware/get-started/adk-install) as startingpoint.
install a minimal version of Windows ADK, then install the Windows PE add-on for the ADK on top. this add-on includes the basic WinPE you can extend and create an ISO from it.

for more detailed explainations/documentations i don't find the time to it... sorry.
but i keep that issue open...

squadramunter, 2019-07-17:
I found this tutorial that can help you to set this up.
https://ahmermansoor.blogspot.com/2018/11/configure-centos-7-pxe-server-install-windows-10.html

mdiharce, 2019-07-18:
Perfect!!!! 
Thanks to both!!!! :)

closed 2019-09-07 by beta-tester

beta-tester, 2019-10-14:
maybe take a look to [PE Bakery](https://github.com/pebakery/pebakery)