beta-tester/RPi-PXE-Server #24: Feature Request. Make a list with configurable settings for keyboard layout and language settings

squadramunter, 2019-07-17:
Hello I have difficulties setting the right keyboard layout. Can You add English US as second option marked out?

beta-tester, 2019-07-18:
you can adjust the "CUSTOM_..." variables in p2-include.var.sh

beta-tester, 2019-07-26:
PS.: not all distros allows to change the keyboard layout/language settings.
best supported is Debian/Ubuntu. all otheres are very limited in their boot options or even don't have the ability in their Live OS.
maybe you can handle it in the preseed or kickstart scripts.
please read the manual of the distros.

closed 2019-07-26 by beta-tester

beta-tester, 2019-07-26:
PPS.: English US keyboard layout is maybe the easiest layout to get...
remove all the stuff behind the double dashes "--" in the boot options (**p2-include-menu.sh**). that will fall back the settings to its default... that is in the most cases en-US.