beta-tester/RPi-PXE-Server #25: Add POP!_OS to the list

squadramunter, 2019-07-17:
Since POP!_OS has Nvidia drivers preinstalled this is very nice to use steam on a live PXE boot. Nvidia works even using live version. So streaming games is no problem at all.

I like to see this distro in the list be added soon.

beta-tester, 2019-07-26:
didn't got response... couldn't find a link to a LiveDVD/ISO by myself.

squadramunter, 2019-09-14:
Are you kidding me?

Just google pop!_os iso and you go to this page.

https://system76.com/pop

https://pop-iso.sfo2.cdn.digitaloceanspaces.com/19.04/amd64/nvidia/10/pop-os_19.04_amd64_nvidia_10.iso

https://pop-iso.sfo2.cdn.digitaloceanspaces.com/19.04/amd64/intel/10/pop-os_19.04_amd64_intel_10.iso

beta-tester, 2019-09-15:
just give me the information, when you want something...

beta-tester, 2019-09-15:
it is just an ubuntu clone with customizations...
it seems like it isn't a Live system, because it immediately try to start the install application - no Try or Install choice...
but here the settings you need, if you want to add to your server.

```
diff --git a/p2-include-handle.sh b/p2-include-handle.sh
index 49b5eb6..3fa1055 100644
--- a/p2-include-handle.sh
+++ b/p2-include-handle.sh
@@ -52,6 +52,8 @@ _unhandle_iso  $UBUNTU_LTS_X86  $UBUNTU_LTS_X86_URL;
 #_unhandle_iso  $UBUNTU_STUDIO_DAILY_X64  $UBUNTU_STUDIO_DAILY_X64_URL  timestamping;
 _unhandle_iso  $UBUNTU_STUDIO_X64  $UBUNTU_STUDIO_X64_URL;
 handle_iso  $UBUNTU_X64  $UBUNTU_X64_URL;
+handle_iso  $POP_OS_NV_X64  $POP_OS_NV_X64_URL;
+handle_iso  $POP_OS_IN_X64  $POP_OS_IN_X64_URL;
 
 #custom#
 handle_iso  $DESINFECT_X64  $DESINFECT_X64_URL;
diff --git a/p2-include-menu.sh b/p2-include-menu.sh
index 170aa69..3d2677f 100644
--- a/p2-include-menu.sh
+++ b/p2-include-menu.sh
@@ -710,6 +710,44 @@ fi
 ##########################################################################
 #custom#
 
+#========== BEGIN ==========
+if [ -f "$FILE_MENU" ] \
+&& [ -f "$DST_NFS_ETH0/$POP_OS_NV_X64/casper/vmlinuz.efi" ]; then
+    echo  -e "\e[36m    add $POP_OS_NV_X64\e[0m";
+    sudo sh -c "cat << EOF  >> $FILE_MENU
+    ########################################
+    LABEL $POP_OS_NV_X64
+        MENU LABEL pop!_os nVidia x64
+        KERNEL $FILE_BASE$NFS_ETH0/$POP_OS_NV_X64/casper/vmlinuz.efi
+        INITRD $FILE_BASE$NFS_ETH0/$POP_OS_NV_X64/casper/initrd.gz
+        APPEND nfsroot=$IP_ETH0:$DST_NFS_ETH0/$POP_OS_NV_X64 ro netboot=nfs boot=casper hostname=pop-os username=pop-os noprompt ---
+        TEXT HELP
+            Boot to pop!_os nVidia x64 Live
+            User: pop-os
+        ENDTEXT
+EOF";
+fi
+#=========== END ===========
+
+#========== BEGIN ==========
+if [ -f "$FILE_MENU" ] \
+&& [ -f "$DST_NFS_ETH0/$POP_OS_IN_X64/casper/vmlinuz.efi" ]; then
+    echo  -e "\e[36m    add $POP_OS_IN_X64\e[0m";
+    sudo sh -c "cat << EOF  >> $FILE_MENU
+    ########################################
+    LABEL $POP_OS_IN_X64
+        MENU LABEL pop!_os intel x64
+        KERNEL $FILE_BASE$NFS_ETH0/$POP_OS_IN_X64/casper/vmlinuz.efi
+        INITRD $FILE_BASE$NFS_ETH0/$POP_OS_IN_X64/casper/initrd.gz
+        APPEND nfsroot=$IP_ETH0:$DST_NFS_ETH0/$POP_OS_IN_X64 ro netboot=nfs boot=casper hostname=pop-os username=pop-os noprompt ---
+        TEXT HELP
+            Boot to pop!_os intel x64 Live
+            User: pop-os
+        ENDTEXT
+EOF";
+fi
+#=========== END ===========
+
 #========== BEGIN ==========
 if [ -f "$FILE_MENU" ] \
 && [ -f "$DST_NFS_ETH0/$DESINFECT_X86/casper/vmlinuz" ]; then
diff --git a/p2-include-url.sh b/p2-include-url.sh
index 32862ca..6326dd2 100644
--- a/p2-include-url.sh
+++ b/p2-include-url.sh
@@ -177,6 +177,12 @@ UBUNTU_X64_URL=http://releases.ubuntu.com/19.04/ubuntu-19.04-desktop-amd64.iso
 UBUNTU_X64_HASH_TYPE=sha256
 UBUNTU_X64_HASH_PATTERN=SHA256SUMS
 
+POP_OS_NV_X64=pop-os-nvidia-x64
+POP_OS_NV_X64_URL=https://pop-iso.sfo2.cdn.digitaloceanspaces.com/19.04/amd64/nvidia/10/pop-os_19.04_amd64_nvidia_10.iso
+
+POP_OS_IN_X64=pop-os-intel-x64
+POP_OS_IN_X64_URL=https://pop-iso.sfo2.cdn.digitaloceanspaces.com/19.04/amd64/intel/10/pop-os_19.04_amd64_intel_10.iso
+
 
 #custom#
 DESINFECT_X64=desinfect-x64

```

closed 2019-09-15 by beta-tester