beta-tester/RPi-PXE-Server #28: RPi PXE Server and Boot from UPBoard (UPC1DM11)

SoloAstra, 2019-12-11:
I tried boot UPBoard with BIOS (UPC1DM11)

>>Checking Media Presence.........
>>Media Present...........
>>Start PXE over IPv4/IPv6
    PXE-E18: Server response timeout.

I tried boot in VirtualBox (Bridged, NAT)

Nothing to boot: No such file or directory (http://ipxe.org/2d03e13b) / Permission denied (http://ipxe.org/0227603c)

I run scripts on RPi 2 with UPBoard using LAN

beta-tester, 2019-12-12:
hi,

can you take a look if all required services are up and running
```
sudo systemctl status dnsmasq
sudo systemctl status nfs-kernel-server
sudo systemctl status lighttpd
sudo systemctl status smbd
```

and if during booting the PC you can observe DHCP logs:
```
tail -F /var/log/syslog
```
during booting your PC, you should see something like:
```
dnsmasq-dhcp[.]:  available DHCP range: 192.168.1.200 -- 192.168.1.250
dnsmasq-dhcp[.]:  client provides name: XXXXXXXX
dnsmasq-dhcp[.]:  available DHCP range: 192.168.1.200 -- 192.168.1.250
dnsmasq-dhcp[.]:  vendor class: PXEClient:Arch:00000:UNDI:002001
dnsmasq-dhcp[.]:  DHCPDISCOVER(eth0) XX:XX:XX:XX:XX:XX
dnsmasq-dhcp[.]:  tags: x86_BIOS, ARCH_0, eth0
dnsmasq-dhcp[.]:  DHCPOFFER(eth0) 192.168.1.217 XX:XX:XX:XX:XX:XX
dnsmasq-dhcp[.]:  requested options: 1:netmask, 2:time-offset, 3:router, 4, 5,
dnsmasq-dhcp[.]:  requested options: 6:dns-server, 11, 12:hostname, 13:boot-file-size,
dnsmasq-dhcp[.]:  requested options: 15:domain-name, 16:swap-server, 17:root-path,
dnsmasq-dhcp[.]:  requested options: 18:extension-path, 22:max-datagram-reassembly,
dnsmasq-dhcp[.]:  requested options: 23:default-ttl, 28:broadcast, 40:nis-domain,
dnsmasq-dhcp[.]:  requested options: 41:nis-server, 42:ntp-server, 43:vendor-encap,
dnsmasq-dhcp[.]:  requested options: 50:requested-address, 51:lease-time, 54:server-identifier,
dnsmasq-dhcp[.]:  requested options: 58:T1, 59:T2, 60:vendor-class, 66:tftp-server,
dnsmasq-dhcp[.]:  requested options: 67:bootfile-name, 128, 129, 130, 131, 132,
dnsmasq-dhcp[.]:  requested options: 133, 134, 135
dnsmasq-dhcp[.]:  next server: 192.168.1.100
dnsmasq-dhcp[.]:  broadcast response
dnsmasq-dhcp[.]:  sent size:  1 option: 53 message-type  2
dnsmasq-dhcp[.]:  sent size:  4 option: 54 server-identifier  192.168.1.100
dnsmasq-dhcp[.]:  sent size:  4 option: 51 lease-time  1d
dnsmasq-dhcp[.]:  sent size: 22 option: 67 bootfile-name  menu-bios/lpxelinux.0
dnsmasq-dhcp[.]:  sent size:  4 option: 58 T1  12h
dnsmasq-dhcp[.]:  sent size:  4 option: 59 T2  21h
dnsmasq-dhcp[.]:  sent size:  4 option:  1 netmask  255.255.255.0
dnsmasq-dhcp[.]:  sent size:  4 option: 28 broadcast  192.168.1.255
dnsmasq-dhcp[.]:  sent size:  4 option:  3 router  192.168.1.100
dnsmasq-dhcp[.]:  sent size:  4 option:  6 dns-server  192.168.1.100
dnsmasq-dhcp[.]:  sent size:  4 option: 42 ntp-server  192.168.1.100
dnsmasq-dhcp[.]:  sent size:  8 option: 66 tftp-server  0.0.0.0
dnsmasq-dhcp[.]:  sent size:  9 option: 60 vendor-class  XX:XX:XX:XX:XX:XX:XX:XX:XX
dnsmasq-dhcp[.]:  sent size: 17 option: 97 client-machine-id  XX:XX:XX:XX:XX:XX:XX:XX:XX
dnsmasq-dhcp[.]:  sent size: 50 option: 43 vendor-encap  XX:XX:XX:XX:XX:XX:XX:XX:XX
dnsmasq-dhcp[.]:  available DHCP range: 192.168.1.200 -- 192.168.1.250
dnsmasq-dhcp[.]:  vendor class: PXEClient:Arch:00000:UNDI:002001
dnsmasq-dhcp[.]:  DHCPDISCOVER(eth0) XX:XX:XX:XX:XX:XX
dnsmasq-dhcp[.]:  tags: x86_BIOS, ARCH_0, eth0
dnsmasq-dhcp[.]:  DHCPOFFER(eth0) 192.168.1.217 XX:XX:XX:XX:XX:XX
dnsmasq-dhcp[.]:  requested options: 1:netmask, 2:time-offset, 3:router, 4, 5,
dnsmasq-dhcp[.]:  requested options: 6:dns-server, 11, 12:hostname, 13:boot-file-size,
dnsmasq-dhcp[.]:  requested options: 15:domain-name, 16:swap-server, 17:root-path,
dnsmasq-dhcp[.]:  requested options: 18:extension-path, 22:max-datagram-reassembly,
dnsmasq-dhcp[.]:  requested options: 23:default-ttl, 28:broadcast, 40:nis-domain,
dnsmasq-dhcp[.]:  requested options: 41:nis-server, 42:ntp-server, 43:vendor-encap,
dnsmasq-dhcp[.]:  requested options: 50:requested-address, 51:lease-time, 54:server-identifier,
dnsmasq-dhcp[.]:  requested options: 58:T1, 59:T2, 60:vendor-class, 66:tftp-server,
dnsmasq-dhcp[.]:  requested options: 67:bootfile-name, 128, 129, 130, 131, 132,
dnsmasq-dhcp[.]:  requested options: 133, 134, 135
dnsmasq-dhcp[.]:  next server: 192.168.1.100
dnsmasq-dhcp[.]:  broadcast response
dnsmasq-dhcp[.]:  sent size:  1 option: 53 message-type  2
dnsmasq-dhcp[.]:  sent size:  4 option: 54 server-identifier  192.168.1.100
dnsmasq-dhcp[.]:  sent size:  4 option: 51 lease-time  1d
[... 58 lines cut ...]
dnsmasq-tftp[.]: sent /srv/tftp/menu-bios/lpxelinux.0 to 192.168.1.217
dnsmasq-tftp[.]: sent /srv/tftp/menu-bios/ldlinux.c32 to 192.168.1.217
dnsmasq-tftp[.]: file /srv/tftp/menu-bios/pxelinux.cfg/xxxxxxxx-xxxx-xxxx-xxxx-xxxxxxxxxxxx not found
dnsmasq-tftp[.]: file /srv/tftp/menu-bios/pxelinux.cfg/XX-XX-XX-XX-XX-XX-XX not found
dnsmasq-tftp[.]: file /srv/tftp/menu-bios/pxelinux.cfg/XXXXXXXX not found
dnsmasq-tftp[.]: file /srv/tftp/menu-bios/pxelinux.cfg/XXXXXXX not found
dnsmasq-tftp[.]: file /srv/tftp/menu-bios/pxelinux.cfg/XXXXXX not found
dnsmasq-tftp[.]: file /srv/tftp/menu-bios/pxelinux.cfg/XXXXX not found
dnsmasq-tftp[.]: file /srv/tftp/menu-bios/pxelinux.cfg/XXXX not found
dnsmasq-tftp[.]: file /srv/tftp/menu-bios/pxelinux.cfg/XXX not found
dnsmasq-tftp[.]: file /srv/tftp/menu-bios/pxelinux.cfg/C0 not found
dnsmasq-tftp[.]: file /srv/tftp/menu-bios/pxelinux.cfg/C not found
dnsmasq-tftp[.]: sent /srv/tftp/menu-bios/pxelinux.cfg/default to 192.168.1.217
dnsmasq-tftp[.]: sent /srv/tftp/menu-bios/vesamenu.c32 to 192.168.1.217
dnsmasq-tftp[.]: sent /srv/tftp/menu-bios/libcom32.c32 to 192.168.1.217
dnsmasq-tftp[.]: sent /srv/tftp/menu-bios/libutil.c32 to 192.168.1.217
dnsmasq-tftp[.]: sent /srv/tftp/menu-bios/pxelinux.cfg/default to 192.168.1.217
```

is your PC directly connected to your router?
in some cases it looks that the network ports on the router are "eating" the DHCP/PXE packets.
see #15.
there is was necessary to plug the PC and the RPi to a switch instead of to the router.
you also can plug the RPi directly to the PC by using a normal standard network cable
("modern" PC's can handle that without using a crosover network cable)

beta-tester, 2020-01-06:
got no response...
i close this issue.

closed 2020-01-06 by beta-tester