beta-tester/RPi-PXE-Server #31: Tails isn't PXE-booting; missing net modules

beta-tester, 2020-04-16:
Tails 4.5 isn't PXE booting, because of missing network modules.
**/lib/modules/5.4.0-4-amd64/kernel/net/** folder is missing **ethernet** and **phy** modules.

beta-tester, 2020-04-16:
follow these steps to create a `tails-x64-hotfix-pxe.cpio.xz` file that contains the network modules and a patch.

1.  in `p2-include-handle.sh` change
`_unhandle_iso  $TAILS_X64  $TAILS_X64_URL;` to
`handle_iso  $TAILS_X64  $TAILS_X64_URL;`.
2. run `install-pxe-server_pass2.sh` to get `tails-x64.iso` downloaded and mounted.
3. be sure, you have the packages `squashfs-tools`, `initramfs-tools` and `xz-utils` installed.
4. modify the following script to the actual tails iso content.
5. execute the following script to create the file `tails-x64-hotfix-pxe.cpio.xz` that will contain the missing network modules and a patch:
```bash
#!/bin/sh

# requires:
#   squashfs-tools  (unsquashfs)
#   initramfs-tools (cpio)
#   xz-utils        (xz)

# location, where to store temporary files
TMP=/tmp/tails-net

# full filename of the filesystem.squashfs from tails ISO
SRC=/srv/nfs/tails-x64/live/filesystem.squashfs

# full filename of the hotfix-pxe image
DST=/srv/nfs/tails-x64-hotfix-pxe.cpio.xz

# kernel version of tails
KVER=5.4.0-4-amd64

# extract missing network kernel drivers modules from tails
sudo unsquashfs \
    -d $TMP \
    -f $SRC \
    -e /lib/modules/$KVER/kernel/drivers/net/phy \
    -e /lib/modules/$KVER/kernel/drivers/net/ethernet \
    ;

# align the modules to the layout of initrd
sudo mkdir $TMP/usr/
sudo mv $TMP/lib $TMP/usr/ 

# add hotfix for pxe boot to initrd image 
sudo mkdir -p $TMP/conf/conf.d/
cat << EOF | sudo tee $TMP/conf/conf.d/1000-hotfix-pxe &>/dev/null
#!/bin/sh

# comment out all blacklist entries
sed "s/^install/# install/g" -i /etc/modprobe.d/all-net-blacklist.conf

# replace wget script by busybox, for normal behavior
ln -sf /bin/busybox /usr/bin/wget

# replace depmod, for normal behavior
ln -sf /bin/kmod /usr/sbin/depmod

# rebulid dependencies for added network kernel drivers modules 
depmod -b /usr

# append hotfix-pxe patch to live script
cat << EOT >> /scripts/live

local_bottom ()
{
    # hotfix-pxe for issue with network initialisation in tails
    local device_
    for device_ in /sys/class/net/*; do
        local device
        device=\\\$(basename \\\$device_)
        if [ "\\\$device" != "lo" ]; then
            # set network devices down
            ip link set \\\$device down

            local module_
            module_=\\\$(readlink \\\$device_/device/driver/module)
            if [ -n "\\\$module_" ]; then
                # remove used network drivers
                local module
                module=\\\$(basename \\\$module_)
                modprobe -r \\\$module 
            fi
        fi
    done
}
EOT
EOF
sudo chmod +x $TMP/conf/conf.d/1000-hotfix-pxe

# create an initrd image to overlay at boot time 
sudo rm $DST
cd $TMP
find . -type f -print0 | cpio --null --create --verbose --format=newc \
    | xz --compress --extreme --check=crc32 | sudo tee $DST &>/dev/null
cd -

# clean up temporary files
sudo rm -rf $TMP

```

beta-tester, 2020-06-30:
procedure not working for tails 4.8.
computer turns off somewhere at initialization.

beta-tester, 2020-08-02:
with tails 4.9, same issue as with tails 4.8.
computer turns off after late initialization.

beta-tester, 2021-02-09:
masking a tails service did the trick.
`systemd.mask=tails-shutdown-on-media-removal.service`

beta-tester, 2021-11-07:
fix for tails 4.24:
ISSUE: overlaying net drivers into place in initrd.img isn't possible anymore for unknown reason.
WORKAROUND: putting net drivers in a net_drivers.tar.xz file and extracting them into place at running boot-scripts.
new script available at https://github.com/beta-tester/RPi-PXE-Server/issues/31#issuecomment-614438868

beta-tester, 2021-11-07:
tails 4.24 doesn't pxe boot anymore. shows same behavior as booting without the net driver patch.

beta-tester, 2021-11-07:
tails 4.24doesn't pxe boot anymore. shows same behavior as booting without net driver patch.

modrz44, 2023-08-19:
This patch stopped working starting I think was Tails 6.10.
I couldn't get an error code, it just refuses to boot once the initrd is download from tftp without any error.
I believe the problem started when Tails switched to the Linux 6.x kernel. My best analysis is that its not the actual patch thats the problem but Linux 6.x kernels handle initrd images differently somehow which makes it refuse any appending initrd patches like thisone.
I've been trying for a few weeks but got nowhere. Hopefully someone can at least establish why this happens.

beta-tester, 2023-08-25:
for me it is working.
- i used Tails tails-amd64-5.16.1.iso
- and applied the script above.
- and PXE boot into Tails (no SecureBoot, no UEFI - just good old BIOS PXE boot)

the kernel version of Tails 5.16.1 is 6.1.0-11-amd64

beta-tester, 2023-09-11:
i changed the script to keep network alive for debugging, if boot any "`break`" option is given
e.g.: "`break=init debug --`"

beta-tester, 2024-02-27:
fix for tails 6.0:
- renamed to all-net-blocklist.conf
- updated path from /lib/modules/ to /usr/lib/modules/
- modified patch-top and patch_bottom

fixed the script above
and added path-tails.sh to project. script still has to be executed manually to apply the patch.

beta-tester, 2025-09-26:
fix for tails 7.0
see commit https://github.com/beta-tester/RPi-PXE-Server/commit/ce5084a41d3e1a93d981a94311aeb548e094e119

closed 2025-09-26 by beta-tester