beta-tester/RPi-PXE-Server #32: Help wanted with Bios selection

jadams3710, 2020-08-07:
I have an older eeePC I am trying to use as lab node to PXE boot 
So far it gets the IP and then TFTP
PXE-01: File not Found
PXE-E3B: TFTP Error - File not found
PXE-M0F: Exiting 

On my Raspberry PI - My host with RPi-PXE-Server installed
syslog has following
Aug  7 13:34:23 ubuntu in.tftpd[4158]: RRQ from 192.168.2.204 filename menu-bios/lpxelinux.0
Aug  7 13:34:27 ubuntu dnsmasq-dhcp[1400]: 1736481460 available DHCP range: 192.168.2.200 -- 192.168.2.250
Aug  7 13:34:27 ubuntu dnsmasq-dhcp[1400]: 1736481460 vendor class: android-dhcp-7.0
Aug  7 13:34:27 ubuntu dnsmasq-dhcp[1400]: 1736481460 client provides name: android-9d9229f139968a96

Where do I set a different bios setting? 
Is there a generic option that will just give a temp option so that I can get menu?

Thanks
And great scripts made for easy setup

beta-tester, 2020-08-07:
hello...
you have a Raspberry Pi as pxe-server and try to pxe-boot your eeePC?
funnyly i have an old eeePC 1000H - with broken battery - but it is pxe-booting a x86 system from my Raspberry Pi

so you get this on your eeePC:
```
PXE-01: File not Found
PXE-E3B: TFTP Error - File not found
PXE-M0F: Exiting
```
on my eeePC i get:
```
Intel UNDI, PXE-2.1 (build 082)
Copyright (C) 1997-2000  Intel Corporation

For Atheros AR8121/AR8113 PCIE Ethernet Controller v1.0.0.4 (2008/01/13)

CLIENT MAC ADDR: XX XX XX XX XX XX  GUID: XXXXXXXX-XXXX-XXXX-XXXX-XXXXXXXXXXXX

DHCP...-/|\-/|\-/|\-/|\

CLIENT IP: 192.168.XXX.YYY  MASK: 255.255.255.0  DHCP IP: 192.168.XXX.ZZZ
GATEWAY IP: 192.168.XXX.ZZZ

PXELINUX 6.04 lwIP 20190226 Copyright (C) 1994-2015 H. Peter Anvin et al
```
and then the pxe menu is shown...

and you don't reach that menu - is that right?

- try to connect your eeePC directly to network port of the pxe-server, without a switch/router in between.
- try to replace the IP 0.0.0.0 by the IP of the pxe-server the file `/etc/dnsmasq.d/10-pxe-server`

what ISO's are in your configuration?

BTW:
where is the "ubuntu" log coming from...
and why there is "android" in the list...?

sorry, i am a little bit confused right now.

jadams3710, 2020-08-07:
So I get the same beginning get IP assigned from my PXE Server 
After the IP assigned I get
Auto-select:
         PXE BootMenu (BIOS 00:00)
Boot server IP:192.168.2.51
TFTP.
PXE-01: File not Found
PXE-E3B: TFTP Error - File not found
PXE-M0F: Exiting Intel PXE ROM.

reboot or Select proper Boot Device.
PXE Menu is never shown after that - 

The log file is from my PXE server 
PXE server is a RPi4 Running Ubuntu 20

I have the following OSs for selection from boot menu
UBUNTU_20_LTS_X64=ubuntu-lts-20-4-live
UBUNTU_16_4_LTS_X86=ubuntu-lts-16-04
UBUNTU_16_10_LTS_X86=ubuntu-lts-16-10
UBUNTU_18_LTS_X64=ubuntu-lts-18-04
UBUNTU_19_LTS_X64=ubuntu-lts-19-10
CENTOS_7_2003_X64=centos-7-2003-min
CENTOS_8_2004_X64=centos-8-2-2004-min

All the menu files get created correctly I have modified the scripts to just download these specific OSs and have all the boot files set

The Android part is missing now.

Latest Log from PXE Server
Aug  7 17:58:36 pxe dnsmasq-dhcp[2032]: 3139550431 broadcast response
Aug  7 17:58:36 pxe dnsmasq-dhcp[2032]: 3139550431 sent size:  1 option: 53 message-type  5
Aug  7 17:58:36 pxe dnsmasq-dhcp[2032]: 3139550431 sent size:  4 option: 54 server-identifier  192.168.2.51
Aug  7 17:58:36 pxe dnsmasq-dhcp[2032]: 3139550431 sent size:  4 option: 51 lease-time  1h
Aug  7 17:58:36 pxe dnsmasq-dhcp[2032]: 3139550431 sent size: 22 option: 67 bootfile-name  menu-bios/lpxelinux.0
Aug  7 17:58:36 pxe dnsmasq-dhcp[2032]: 3139550431 sent size:  4 option: 58 T1  30m
Aug  7 17:58:36 pxe dnsmasq-dhcp[2032]: 3139550431 sent size:  4 option: 59 T2  52m30s
Aug  7 17:58:36 pxe dnsmasq-dhcp[2032]: 3139550431 sent size:  4 option:  1 netmask  255.255.255.0
Aug  7 17:58:36 pxe dnsmasq-dhcp[2032]: 3139550431 sent size:  4 option: 28 broadcast  192.168.2.255
Aug  7 17:58:36 pxe dnsmasq-dhcp[2032]: 3139550431 sent size:  4 option:  3 router  192.168.2.51
Aug  7 17:58:36 pxe dnsmasq-dhcp[2032]: 3139550431 sent size:  4 option:  6 dns-server  192.168.2.51
Aug  7 17:58:36 pxe dnsmasq-dhcp[2032]: 3139550431 sent size:  9 option: 60 vendor-class  50:58:45:43:6c:69:65:6e:74
Aug  7 17:58:36 pxe dnsmasq-dhcp[2032]: 3139550431 sent size: 17 option: 97 client-machine-id  00:00:a0:70:d5:66:06:46:81:3b:9b:90:e6:ba...
Aug  7 17:58:36 pxe dnsmasq-dhcp[2032]: 3139550431 sent size: 50 option: 43 vendor-encap  06:01:03:0a:04:00:50:58:45:08:07:80:01:01...
Aug  7 17:58:36 pxe dnsmasq-dhcp[2032]: 3139550431 available DHCP range: 192.168.2.200 -- 192.168.2.250
Aug  7 17:58:36 pxe dnsmasq-dhcp[2032]: 3139550431 vendor class: PXEClient:Arch:00000:UNDI:002001
Aug  7 17:58:36 pxe dnsmasq-dhcp[2032]: 3139550431 PXE(eth0) 192.168.2.204 90:e6:ba:21:bc:df menu-bios/lpxelinux.0
Aug  7 17:58:36 pxe dnsmasq-dhcp[2032]: 3139550431 tags: x86_BIOS, ARCH_0, eth0
Aug  7 17:58:36 pxe dnsmasq-dhcp[2032]: 3139550431 bootfile name: menu-bios/lpxelinux.0
Aug  7 17:58:36 pxe dnsmasq-dhcp[2032]: 3139550431 next server: 192.168.2.51
Aug  7 17:58:36 pxe dnsmasq-dhcp[2032]: 3139550431 sent size:  1 option: 53 message-type  5
Aug  7 17:58:36 pxe dnsmasq-dhcp[2032]: 3139550431 sent size:  4 option: 54 server-identifier  192.168.2.51
Aug  7 17:58:36 pxe dnsmasq-dhcp[2032]: 3139550431 sent size:  9 option: 60 vendor-class  50:58:45:43:6c:69:65:6e:74
Aug  7 17:58:36 pxe dnsmasq-dhcp[2032]: 3139550431 sent size: 17 option: 97 client-machine-id  00:00:a0:70:d5:66:06:46:81:3b:9b:90:e6:ba...
Aug  7 17:58:36 pxe dnsmasq-dhcp[2032]: 3139550431 sent size:  7 option: 43 vendor-encap  47:04:80:01:00:00:ff
Aug  7 17:58:37 pxe in.tftpd[2052]: RRQ from 192.168.2.204 filename menu-bios/lpxelinux.0
Aug  7 17:58:37 pxe in.tftpd[2053]: RRQ from 192.168.2.204 filename menu-bios/lpxelinux.0
Aug  7 17:59:39 pxe dnsmasq-dhcp[2032]: 1880368748 available DHCP range: 192.168.2.200 -- 192.168.2.250
Aug  7 17:59:42 pxe dnsmasq-dhcp[2032]: 1880368748 DHCPDISCOVER(eth0) 98:41:5c:40:55:eb 
Aug  7 17:59:42 pxe dnsmasq-dhcp[2032]: 1880368748 tags: eth0
Aug  7 17:59:42 pxe dnsmasq-dhcp[2032]: 1880368748 DHCPOFFER(eth0) 192.168.2.238 98:41:5c:40:55:eb 
Aug  7 17:59:42 pxe dnsmasq-dhcp[2032]: 1880368748 requested options: 1:netmask, 3:router, 6:dns-server, 28:broadcast
Aug  7 17:59:42 pxe dnsmasq-dhcp[2032]: 1880368748 server name: 192.168.2.51
Aug  7 17:59:42 pxe dnsmasq-dhcp[2032]: 1880368748 next server: 192.168.2.51
Aug  7 17:59:42 pxe dnsmasq-dhcp[2032]: 1880368748 sent size:  1 option: 53 message-type  2

Thanks

beta-tester, 2020-08-07:
you have Ubuntu running on the RPi and not Raspberry Pi OS this i didn't expected.
in the log you have the entry 
`pxe in.tftpd[2052]: RRQ from 192.168.2.204 filename menu-bios/lpxelinux.0`
do you have installed tftpd by hand? in my configuration dnsmasq is serving the tftp service.
maybe your tftpd is trying to handle the request from the eeePC... but tftpd is not configured by my script.
try to turn off the tftpd service temporarily and see if you can pxe boot after that
`sudo systemctl stop tftpd.service`

jadams3710, 2020-08-07:
That did it. I must have had the tftpd service installed from one of my other attempts and just didn't get rid of it. 

Thanks
Much for the help greatly appreciated.

beta-tester, 2020-08-07:
ok. i close this issue.

closed 2020-08-07 by beta-tester