beta-tester/RPi-PXE-Server #33: I've added and altered some features

jake5253, 2020-12-10:
I have added and altered some features of your scripts.  If interested, please let me know and I could pull request.
Feautes/alterations are:
* argument to run 'menu-only mode'. -- This runs only handle_pxe. I'm impatient and I see no reason to re-write config scripts again and again if they haven't changed, like dnsmasq config, samba config, etc.
* argument to 'update urls' (and rebuild menu) -- this uses wget to pull only your latest urls file
* include custom -- added 3 new files for custom entries. This allows the update urls to work without losing any customizations
* use local ISOs -- uses an is_local flag to declare that an ISO is already present on the system, for instance, if you've customized an ISO before you add it to your custom's scripts.  It uses rsync instead of wget since wget gets upset about local file paths when it wants a url.
* disable DHCP server take-over -- I was having trouble when pairing this script with OpenWRT.  They seemed to be fighting over who was going to be the main DHCP server. This setting adds a declared IP as the DHCP server (dhcp option 3) and sets DNS server (dhcp option 6)
* option to disable backups -- I ran out of space on my flash drive and really don't need a backup of ISOs, so this just disables the final 2 backups in pass2.sh

beta-tester, 2020-12-17:
thank you for your hint.
no need for a pull request, i will watch your implementation at your forks
https://github.com/jake5253/rpi-pxe
and
https://github.com/jake5253/RPi-iPXE-Server
and will see, if i pick up some inspiration from you.

jake5253, 2020-12-17:
Certainly, sir! I'm currently using the altered pxe version. I don't believe I've made any uncommited changes to that, other than actually adding custom entries, but I purposely omitted those to prevent any confusion and to eliminate people using old distributions if/when I forget to update my customizations.
The ipxe one is very very-early stage. I'm also playing with the idea of writing from scratch, a fully PHP and sqlite-based solution with authentication and user/machine-based boot options, so I haven't done many alterations to the ipxe variant of your work, and ultimately I might ditch the ipxe-variant. I might look for a way to expand the PHP-ipxe to allow users to use either menu.
Meanwhile, I have also been looking into using pxe or ipxe to kickstart a GRUB boot menu because I've used GRUB way more and the menus are "more attractive".  So many ideas.  
Also, I'd like to say thank you for sharing your project!  Very cool!

beta-tester, 2021-02-01:
see also #36

closed 2021-02-01 by beta-tester