beta-tester/RPi-PXE-Server #34: trying to use own iso's please can some one help

adambro12, 2020-12-21:
hi can someone please help me as im trying to get my own iso's in the boot i cant get them in to folder and hange boot menu please can someone help or do step to step please and happy holadays

beta-tester, 2020-12-21:
you created your own linux/windows ISO or you took an existing linux distro, that is not in my script included?

- if it is an existinf distro, please give me a link to the webpage, maybe i can add it to my script
- if it is your own custom ISO, what OS ist it (linux or windows), how big is the ISO does it support PXE boot, ...

adambro12, 2020-12-22:
Not one I made its existing OS is like Windows install 7 and 10 and hired
boot and active boot I've been thinking about making a pxe server for years
and you script was just so nice and easy but I was trying to work out if I
could put other iso or img in it
Thank you for getting back to me
Kind regards
Adam

beta-tester, 2020-12-22:
with the Windows ISO media there are two problems.
- it can not fully loaded as RAMdisk into the RAM
- even your computer has enough RAM available, the RAMDisk once it is in the RAM loaded, the setup of windows can not see the content of the ISO inside the RAM.

the way you can install windows over the network is to create a WinPE ISO (~400MB) that fully fits into the RAM.
once the client has pxe booted the windows PE system, it can activate its network.
and mount the share where your the windows 10 installation files are stored.
something like:
```
wpeutil InitializeNetwork
net use N: \\pxe-server\srv guest /USER:guest
N:
cd windows10_install_files
setup.exe
```

for more information:
[Download and install the Windows ADK](https://docs.microsoft.com/de-de/windows-hardware/get-started/adk-install) to create you own Windows PE iso. you nead at least the "Windows PE add-on for the ADK" to biuld your WinPE iso.
[Windows PE (WinPE)](https://docs.microsoft.com/en-us/windows-hardware/manufacture/desktop/winpe-intro)

in the script i have already a template for win-pe-x86 that will create a pxe menu entry as soon there is a ISO called win-pe-x86.iso available

PS.: you only can run the setupof a x64, if your Win-PE is also x64. so you have to create a x64 (amd64) ISO in case you want to install a windows 64bit version.

beta-tester, 2020-12-22:
i added a tiny wiki that shows the steps to create a WinPE ISO
https://github.com/beta-tester/RPi-PXE-Server/wiki/Create-a-WinPE-ISO
and a video https://github.com/beta-tester/test/blob/master/Create_WinPE.mp4?raw=true

adambro12, 2020-12-22:
Tha k you I will have a look when I get home happy holidays

adambro12, 2020-12-22:
wow thank you got winpe working learnt something new and can use tftp for
in stall now would this work https://www.aioboot.com/en/hirens-bootcd-pe/

adambro12, 2020-12-22:
https://www.hirensbootcd.org/old-versions/

closed 2021-01-05 by beta-tester