beta-tester/RPi-PXE-Server #35: Problem booting Windows 10 from Raspberry PXE server

iskopa, 2021-01-03:
Hello,

I am trying a variation of your PXE boot server procedure to boot Windows 10 from the RPi-PXE-Server (https://linuxconfig.org/how-to-configure-a-raspberry-pi-as-a-pxe-boot-server): 

   1.  I have extended the “../pxelinux.cfg/default” as it follows:

    MENU BEGIN Windows

     MENU TITLE Windows 10 amd64

         LABEL windows_10_PE
             MENU LABEL Windows PE (64 Bit)
             KERNEL ::memdisk
             INITRD ::win-pe-x64.iso
             APPEND iso raw            

         MENU END

    2. In order to enable booting in BIOS and EFI modus, I created the following directory structure:
    
/mnt/data/netboot
│win-pe-x64.iso
│memdisk
├── bios
│   ├── ldlinux.c32
│   ├── libcom32.c32
│   ├── libutil.c32
│   ├── pxelinux.0
│   ├── pxelinux.cfg -> ../pxelinux.cfg
│   └── vesamenu.c32
├── boot
│   └── amd64
│       └── windows
│           └── 10
├── efi64
│   ├── ldlinux.e64
│   ├── libcom32.c32
│   ├── libutil.c32
│   ├── pxelinux.cfg -> ../pxelinux.cfg
│   ├── syslinux.efi
│   └── vesamenu.c32
└── pxelinux.cfg
         └── default
     
....and copied the files:

$ cp \
  /usr/lib/syslinux/modules/bios/{ldlinux,vesamenu,libcom32,libutil}.c32 \
  /usr/lib/PXELINUX/pxelinux.0 \
  /mnt/data/netboot/bios

$ cp \
  /usr/lib/syslinux/modules/efi64/ldlinux.e64 \
  /usr/lib/syslinux/modules/efi64/{vesamenu,libcom32,libutil}.c32 \
  /usr/lib/SYSLINUX.EFI/efi64/syslinux.efi \
  /mnt/data/netboot/efi64

3 I then linked the pxelinux.cfg to the "bios" and "efi64" directories
 
     ln -rs pxelinux.cfg bios && ln -rs pxelinux.cfg efi64

4. I linked memdisk to the netboot directory
     ln -s /usr/lib/syslinux/memdisk /mnt/data/netboot/
    
5. I have generated a “win-pe-x64.iso” using microsoft’s “Windows ADK” and “Windows PE add-on” following your tutorial:
 https://github.com/beta-tester/RPi-PXE-Server/wiki/Create-a-WinPE-ISO

6. I placed the “win-pe-x64.iso” file in /mnt/data/netboot

7. I copied a mounted windows iso installation file on a samba directory:

    7.1 sudo mount /home/pi/SambaSharedDirectory/Win10_20H2_v2_EnglishInternational_x64.iso /media
    7.2 sudo rsync -av /media/ /mnt/data/netboot/boot/amd64/windows/10

smb.conf:

[10]
comment = Windows 10 (64 Bit) Install Files
path = /mnt/data/netboot/boot/amd64/windows/10
public = yes
printable = no
guest ok = yes
readonly = yes
browseable = yes
oplocks = no
level2 oplocks = no
locking = no
strict locking = no

8.  dnsmasq.conf 

      port=0 # disable DNS server
      interface=eth0
      dhcp-range=192.168.111.0,proxy

      enable-tftp

      tftp-root=/mnt/data/netboot

      pxe-service=x86PC,"PXELINUX (BIOS)",bios/pxelinux
      pxe-service=x86-64_EFI,"PXELINUX (EFI)",efi64/syslinux.efi

     log-queries
     log-facility=/var/log/dnsmasq.log

9. When I start the windows 10 installation, the client reports the following:

     Loading :: memdisk... ok
     Loading :: win-pe-x64.iso... ok
     **Failed to exit boot services: 0x800000000002
     Booting kernel failed: Bad file number**

Do you know how to solve this problem?

Thank you in advance

beta-tester, 2021-01-03:
when you have a spare SD card, please try my original scripts, to see how this RPi-PXE-Server behaves.

do you know where the syntax description of the "::" in the menu entries is?
i never saw that :: before and can not find any description for that ::

iskopa, 2021-01-03:
Hello and thank your for the quick response.

I would like to stay with my SD installation for the moment, as I have lots of setups running there and would like to extend its features with the PXE server.

If I do not use „::“ in the menu entries, then the client cannot find neither „memdesk“ nor „win-pe-x64.iso“. I found an example of that usage below, see „**Creating the boot menu**“:

https://linuxconfig.org/how-to-configure-a-raspberry-pi-as-a-pxe-boot-server

Best regards

beta-tester, 2021-01-03:
i use the ip address instead of the `::` or more precise i use the http link.
but i guess you don't have a http server running for publish the image files that pxe process is loading after boot menu.
(i use `lpxelinux.0`  instead of `pxelinux.0`, because it supports http for faster loading than tftp)

with the spare SD card  i only wanted to figure out ,if my script behaves the same or not, do know where to look what might be the issue.
my project developed long time ago and i don't remember all the details why i have added some parameters and so on.
so it is hard to tell where exactly your problem comes from, even you added all the details.

can't you ask the creator of the other tutorial you followed?

i don't know if the error message you got is from Windows PE or still from pxelinux...
the samba settings can not be the issue, because the boot process isn't that far at that stage.

do you try to pxe boot from BIOS or UEFI?
i know that i can not pxe boot from UEFI + SecureBoot from that iso in this way with memdisk on my real PC.
i don't know how it is, when SecureBoot isn't enabled - i don't want to mess around with my working PC.
and i con'T test it on VirtualBox, because it does not have an UEFI IPv4 PXE boot option... only BIOS PXE.

iskopa, 2021-01-03:
thank you again,

I deactivated secure boot and tried with both UEFI and BIOS, but unfortunately that did not work. Using the dhcp-boot instructions above did not work either: When the client starts, it cannot load the menu. It reports:
    checking media pressence...
    Media present ....
    Start PXE over IPv4...

and that is all. After this messages the menu should be loaded.
 
Perhaps lpxelinux would solve this, but it looks a long way to go.

beta-tester, 2021-01-03:
i repeat myself, when i say that you can test my scripts on a second SD card...
it would also help me to figure out, if i make things wrong in my scripts/settings.

but if you don't want, then try to PXE the iso of tinyCore
download  http://tinycorelinux.net/11.x/x86_64/release/TinyCorePure64-current.iso it is only about 30MByte.
and try the pxe menu entries
```
LABEL TinyCorePure64
    MENU LABEL TinyCorePure64 (ISO)
    KERNEL ::memdisk
    APPEND iso raw
    INITRD ::TinyCorePure64-current.iso
```
and try to PXE-boot into that iso (at least via BIOS).
will it give tha same error

iskopa, 2021-01-03:
I booted the TinyCore via BIOS and it did work, but not in UEFI modus.
When trying the windows-pe-x64.iso via BIOS, booting stops at this point:
 
[image]

beta-tester, 2021-01-03:
maybe your ISO of winPE is broken. at least pxe boot via BIOS should work.
or what happens when you remove `raw` from the `APPEND iso raw` line? your screenshot looks like it is loaded as a hard drive, and not as an iso image.

iskopa, 2021-01-03:
yes I also changed APPEN to "harddisk" and it could step into the message above. With APPEND = "iso" and "iso raw" it did not get that far.

beta-tester, 2021-01-03:
"harddisk" is definitely not working...
 
strange, that you do not get WinPE booting from BIOS, this should work at least, when you say TinyCore is booting well.

when booting via BIOS would workt but UEFI not, then i would provide you another solution for UEFI, but this makes no sense, then boting via BIOS is not working.

beta-tester, 2021-01-04:
only to be sure, please try `lpxelinux.0`. you should have it already installed on your RPi.
it is compatible to `pxelinux.0` but has some additions.
maybe it behaves different.

`sudo ln -s /usr/lib/PXELINUX/lpxelinux.0 /mnt/data/netboot/bios/`

and in dnsmasq change pxelinux to lpxelinux
```
pxe-service=tag:x86PC, "PXELINUX (BIOS)", bios/lpxelinux
dhcp-boot=tag:x86PC, bios/lpxelinux.0
```

and try to pxe boot via BIOS

beta-tester, 2021-01-04:
here the version with Microsoft's `pxeboot.n12` file:

1. extract the folowing files from your `sources/boot.wim` file of your `win-pe-64.iso`:<br>`pxeboot.n12`, `boormgr.exe`<br>they can be found in the `boot.wim` image under `\Windows\Boot\PXE\`
2. copy `pxeboot.n12` to your server to `/mnt/data/netboot/bios/pxeboot.n12`
3. copy `bootmgr.exe` to your server to `/mnt/data/netboot/bootmgr.exe`
4. extract the folowing folder/subfolders from your `win-pe-64.iso`:<br>`Boot`
5. copy `Boot` to your server to `/mnt/data/netboot/Boot`
6. _make a link from `/mnt/data/netboot/Boot` to `/mnt/data/netboot/boot`<br>(this is only a temporary hack, because Microsoft put a lowercase path in the BCD entries)_
7. add to your `.../bios/pxelinux.cfg/default`the following menu entry:
```
LABEL win-pe-x64-pxe
        MENU LABEL Windows PE x64 (PXE)
        PXE pxeboot.n12
        TEXT HELP
            Boot to Windows PE 64bit via pxeboot.n12
        ENDTEXT
```
try to pxe boot via BIOS.

PS.: note i obmitted the `::`, because the `pxeboot.n12` is inside the `bios` folder.

iskopa, 2021-01-04:
I works now!

I linked the lpxelinux and modified the instructions in dnsmasq as you proposed (without "tag:" on the pxe-service)

pxe-service=x86PC, "PXELINUX (BIOS)",bios/lpxelinux
dhcp-boot=tag:x86PC,bios/lpxelinux.0

Then the client could boot and I could mount and use the samba directory containing windows 10:

in client:
net use Z:  \\192.168.111.128\10

run z:\setup.exe

Thank you very much. It makes fun

beta-tester, 2021-01-04:
nice... so i close the issue

closed 2021-01-04 by beta-tester

NiKiZe, 2021-08-28:
How long does it take you on average to get into winpe with this setup?

iskopa, 2021-08-28:
The booting till winPe Takes about a minute