beta-tester/RPi-PXE-Server #38: Some more image to add.

Mightyretro, 2022-09-22:
Hi,

i came by your project and love it so far and i was wondering if its possible to add some more things to your project

Qubes OS
https://www.qubes-os.org/

Ubuntu Server
https://ubuntu.com/download/server

Proxmox VE
https://www.proxmox.com/en/downloads/category/iso-images-pve

XCP-ng
https://xcp-ng.org/#easy-to-install

MemTest86
https://www.memtest86.com/download.htm

MemTest86+
https://www.memtest.org/

Rescuezilla
https://rescuezilla.com/download

boot-repair-disk
https://sourceforge.net/projects/boot-repair-cd/

Darik’s Boot And Nuke
https://dban.org/

Shredos
https://github.com/PartialVolume/shredos.x86_64

ESET SysRescue Live
https://www.eset.com/int/support/sysrescue/#c66026

pfSense
https://www.pfsense.org/download/

TrueNAS Core
https://www.truenas.com/download-truenas-core/

TrueNAS Scale
https://www.truenas.com/download-truenas-scale/

Bliss OS 14.3 FOSS (x86_64)
https://sourceforge.net/projects/blissos-dev/files/Android-Generic/PC/bliss/R/foss/BlissOS-14.3-x86_64-202107242243_k-android12-5.10.50-ax86_m-21.1.3_r-x86_foss_cros-hd.iso/download

Passcape Reset Windows Password 
https://www.passcape.com/reset_windows_password

Passcape Win CD Keys
https://www.passcape.com/windows_cd_keys

Kon-Boot
https://kon-boot.com/

VeraCrypt Rescue
https://veracrypt.eu/en/VeraCrypt%20Rescue%20Disk.html

Kind Regards // Marc

beta-tester, 2022-09-27:
> Qubes OS https://www.qubes-os.org/

maybe, i will see...

> Ubuntu Server https://ubuntu.com/download/server

its only an installer, not a live system

> Proxmox VE https://www.proxmox.com/en/downloads/category/iso-images-pve

no, no static download link

> XCP-ng https://xcp-ng.org/#easy-to-install

maybe, i will see...

> MemTest86 https://www.memtest86.com/download.htm

maybe, i will see...

> MemTest86+ https://www.memtest.org/

maybe, i will see...

> Rescuezilla https://rescuezilla.com/download

definitely on my to-do list

> boot-repair-disk https://sourceforge.net/projects/boot-repair-cd/

maybe, i will see...

> Darik’s Boot And Nuke https://dban.org/

no, needs registration for download

> Shredos https://github.com/PartialVolume/shredos.x86_64

maybe, i will see...

> ESET SysRescue Live https://www.eset.com/int/support/sysrescue/#c66026

is already in

> pfSense https://www.pfsense.org/download/

maybe, i will see...

> TrueNAS Core https://www.truenas.com/download-truenas-core/
> TrueNAS Scale https://www.truenas.com/download-truenas-scale/

no, needs registration for download

> Bliss OS 14.3 FOSS (x86_64) https://sourceforge.net/projects/blissos-dev/files/Android-Generic/PC/bliss/R/foss/BlissOS-14.3-x86_64-202107242243_k-android12-5.10.50-ax86_m-21.1.3_r-x86_foss_cros-hd.iso/download

maybe, i will see...

> Passcape Reset Windows Password https://www.passcape.com/reset_windows_password

no, not free

> Passcape Win CD Keys https://www.passcape.com/windows_cd_keys

no, not free

> Kon-Boot https://kon-boot.com/

no, not free

> VeraCrypt Rescue https://veracrypt.eu/en/VeraCrypt%20Rescue%20Disk.html

rescue disk is too old, no ISO

beta-tester, 2022-11-17:
added Rescuezilla
see commit https://github.com/beta-tester/RPi-PXE-Server/commit/4a701b4dfa266825beca1e375af7e9f43609003d

closed 2022-12-23 by beta-tester