beta-tester/RPi-PXE-Server #39: Use wifi (wlan0) with pxe in setup script (p1-setup)

TinyRobby, 2022-10-03:
so all I really want to know is can you do it

beta-tester, 2022-10-04:
you mean you want to PXE/netboot a computer by using only wifi instead of ethernet cable?

no, not with stock BIOS/firmware.

technically it could be possible.
the BIOS/firmware would require, the knowledge of ssid and its credentials of the access point to get access to the network over wifi.
i don't know any BIOS that supports that kind of PXE/netboot.
maybe there is a special modified BIOS/firmware available somewhere that has the possibility,

TinyRobby, 2022-10-04:
What I mean is when I start the setup script it dosen't continue because there is no ethernet cable connected

beta-tester, 2022-10-04:
oh, sorry, that you mean. this is a case i have not on my _radar_.
does it make sense to install the script to a RPi without a ethernet cable connected, when the PYE/netboot over wifi isn't supported?
even when you change the _eth_ variables to _wlan_, the script will leave the system setup to the wrong interface and ip address, that you have to correct my hand manually.

TinyRobby, 2022-10-06:
I just wanted to know because my router isn't close enough to my desk for me to plug in an ethernet cable. I think I might just move my pi closer.

closed 2022-10-06 by TinyRobby