beta-tester/RPi-PXE-Server #40: Endeavouros_Cassini_Nova-03-2023_R1.iso

Vykeshnaidu, 2023-06-16:


beta-tester, 2023-06-17:
no description...?
what is you issue?

i recently tried 
```bash
MANJARO_X64=manjaro-x64
MANJARO_X64_URL=https://download.manjaro.org/xfce/22.1.0/manjaro-xfce-22.1.0-minimal-230421-linux61.iso

ENDEAVOUR_X64=endeavour-x64
ENDEAVOUR_X64_URL=https://mirror.alpix.eu/endeavouros/iso/EndeavourOS_Artemis-22_6.iso
#https://github.com/endeavouros-team/ISO/releases/download/1-EndeavourOS-ISO-releases-archive/EndeavourOS_Cassini_Nova-03-2023_R1.iso
```

with
```bash
#========== BEGIN ==========
if [[ -f "$FILE_MENU" ]] \
&& [[ -f "$DST_NFS_ETH0/$MANJARO_X64/boot/vmlinuz-x86_64" ]]; then
    echo  -e "\e[36m    add $MANJARO_X64\e[0m";
    cat << EOF | sudo tee -a $FILE_MENU &>/dev/null
    ########################################
    LABEL $MANJARO_X64
        MENU LABEL manjora x64
        KERNEL $FILE_BASE$NFS_ETH0/$MANJARO_X64/boot/vmlinuz-x86_64
        INITRD $FILE_BASE$NFS_ETH0/$MANJARO_X64/boot/intel_ucode.img,$FILE_BASE$NFS_ETH0/$MANJARO_X64/boot/amd_ucode.img,$FILE_BASE$NFS_ETH0/$MANJARO_X64/boot/initramfs-x86_64.img
        #APPEND ip=dhcp ro archisobasedir=manjaro archiso_nfs_srv=$IP_ETH0:$DST_NFS_ETH0/$MANJARO_X64 copytoram=n
        APPEND vmlinuz-x86_64 initrd=initramfs-x86_64.img ip=dhcp ro misolabel=manjaro misobasedir=manjaro miso_nfs_srv=$IP_ETH0:$DST_NFS_ETH0/$MANJARO_X64 copytoram=n
        #SYSAPPEND 3
        TEXT HELP
            Boot to manjora Linux
        ENDTEXT
EOF
fi
#=========== END ===========

#========== BEGIN ==========
if [[ -f "$FILE_MENU" ]] \
&& [[ -f "$DST_NFS_ETH0/$ENDEAVOUR_X64/arch/boot/x86_64/vmlinuz-linux" ]]; then
    echo  -e "\e[36m    add $ENDEAVOUR_X64\e[0m";
    cat << EOF | sudo tee -a $FILE_MENU &>/dev/null
    ########################################
    LABEL $ENDEAVOUR_X64
        MENU LABEL ENDEAVOUR_X64 x64
        KERNEL $FILE_BASE$NFS_ETH0/$ENDEAVOUR_X64/arch/boot/x86_64/vmlinuz-linux
        INITRD $FILE_BASE$NFS_ETH0/$ENDEAVOUR_X64/arch/boot/intel-ucode.img,$FILE_BASE$NFS_ETH0/$ENDEAVOUR_X64/arch/boot/amd-ucode.img,$FILE_BASE$NFS_ETH0/$ENDEAVOUR_X64/arch/boot/x86_64/initramfs-linux.img
        #APPEND ip=dhcp ro archisobasedir=arch archiso_nfs_srv=$IP_ETH0:$DST_NFS_ETH0/$ENDEAVOUR_X64 copytoram=n
        APPEND ip=dhcp ro archisobasedir=arch archiso_http_srv=http://$IP_ETH0$DST_NFS_ETH0/$ENDEAVOUR_X64/ copytoram=n
        #APPEND ip=dhcp ro archisobasedir=arch archiso_nfs_srv=192.168.10.1:/srv/nfs/endeavour-x64 copytoram=n
        #APPEND ip=dhcp ro archisobasedir=arch archiso_http_srv=http://192.168.10.1/srv/nfs/endeavour-x64/ nouveau.modeset=1 module_blacklist=nvidia,nvidia_modeset,nvidia_uvm,nvidia_drm i915.modeset=1 radeon.modeset=1 nvme_load=yes
        #APPEND ip=dhcp ro archisobasedir=arch archiso_nfs_srv=192.168.10.1:/srv/nfs/endeavour-x64 module_blacklist=nvidia,nvidia_modeset,nvidia_uvm,nvidia_drm,i915,radeon,nouveau nomodeset nvme_load=yes debug loglevel=7 logo.nologo
        SYSAPPEND 3
        TEXT HELP
            Boot to ENDEAVOUR_X64 Linux
        ENDTEXT
EOF
fi
#=========== END ===========
```

with no success...

it looks like its live boot iso's is missing http- and nfs- netboot packages in their boot kernel/envoironment.

beta-tester, 2023-06-17:
see hiere:
https://github.com/endeavouros-team/EndeavourOS-ISO/issues/46

closed 2023-06-17 by beta-tester