beta-tester/RPi-PXE-Server #42: PXE-booting patch stopped working starting Tails 6.10.

modrz44, 2023-08-19:
Referring to 
https://github.com/beta-tester/RPi-PXE-Server/issues/31
 This patch stopped working starting I think was Tails 6.10.
I couldn't get an error code, it just refuses to boot once the initrd is download from tftp without any error.
I believe the problem started when Tails switched to the Linux 6.x kernel. My best analysis is that its not the actual patch thats the problem but Linux 6.x kernels handle initrd images differently somehow which makes it refuse any appending initrd patches like thisone.
I've been trying for a few weeks but got nowhere. Hopefully someone can at least establish why this happens.

_Originally posted by @modrz44 in https://github.com/beta-tester/RPi-PXE-Server/issues/31#issuecomment-1684908251_

beta-tester, 2023-08-20:
thank you for the report. i will take a look soon.

beta-tester, 2023-08-25:
for me it is working.

- i used Tails tails-amd64-5.16.1.iso
- and applied the script above.
- and PXE boot into Tails (no SecureBoot, no UEFI - just good old BIOS PXE boot)

the kernel version of Tails 5.16.1 is 6.1.0-11-amd64

beta-tester, 2023-09-11:
i will close this issue, because i got no further response

closed 2023-09-11 by beta-tester