beta-tester/RPi-PXE-Server #43: New Docker implementation

pi-farm, 2023-11-17:
Hello @beta-tester and all other...,
i have build a Docker implementation of this fantastic project.

You may have a look at: 
https://github.com/pi-farm/Docker-PXE-Server/tree/master

Greetings from the Pi-Farm

beta-tester, 2023-11-18:
sounds interesting! :star: :star: :star:

i an not familiar with using Docker :shame:

from time to time i update the `p2-include-url` file and `p2-include-menu` if i notice a required change there.
so a `git pull` may be required to keep the url file up to date.

pi-farm, 2025-03-22:
Greetings from the Pi-Farm, the Docker-PXE-Server has now a tmux based "cli-gui".

Thanks for this great project @beta-tester 

ps: git pull is now implemented 

Greetings from the Pi-Farm